ARM-software/vscode-cmsis-csolution #72: uv2csolution failed

KongJipeng, 2024-07-02:
Type: Bug Report

**Describe the bug**
- OS and Version: Windows11 23H2
- VS Code Version: 1.92.0
- Extension Version: Arm Keil Studio Pack (MDK v6)
- Target Device: STM32F103RCT6
- I can only convert my μVision projec by clicking "Convert a μVision Project" on the initial page of the extension when no project has been opened yet. After opening a project, and I convert my project by "Move your cursor over the Solution outline, click More Actions ···, then select Convert uvproj to csolution".  It will report:
- 
-c:\Users\...\vscode\extensions\arm.cmsis-csolution-1.37.0-win32-x64\tools\uv2csolution\uv2csolution.exe undefined
0 out of 0 target(s) converted
Error: 1 error(s), 0 warning(s):
Errors:
	open undefined: The system cannot find the file specified.
Process failed with exit code 2

But I'm sure the path "c:\user\....\uv2csolution.exe" is right. Did I not configure the correct path somewhere?

[image]

jkrech, 2024-07-02:
Please excuse the confusion caused and thanks for reporting the issue.
The action "Convert a uVision Project" should not show in the `...` menu, while no folder/workspace is open. Also it should not show if no file with `*.uvprojx` or `*.uvmpw` is present there. 
In order for the action to work, the above are mandatory pre-requisites.
Hopefully this should not block your work.

ReinhardKeil, 2024-10-18:
I was unable to reproduce this problem using the VS Code CMSIS solution extension 1.40

closed 2024-10-18 by jkrech