ARM-software/vscode-cmsis-csolution #76: Pack index not updated

KimmoVaisanen-AlifSemi, 2024-10-09:
Type: Bug Report

**Describe the bug**
- OS and Version: Ubuntu 22.04 or Windows 11
- VS Code Version: 1.94.0
- Extension Version: 1.38.0

If new pack is installed into Keil CMSIS pack repository and user updates the pack to a newer version in it's solution. System will try to install the new pack version, but gives an error `E: pack version not found in the pdsc file`.

After debugging and googling the issue, I found that the pdsc file in .cache/arm/packs/.Web is not updated with the latest version therefore resulting in the error.

This can be worked around by running `cpacket update-index`, but should system do this automatically?

If yes, I don't know if this an issue with the vscode-cmsis-csolution extension or the cmsis-toolbox.

jkrech, 2024-10-09:
The current position is that it is not efficient to always update the index.pidx with cpackget commands that access the index.pidx.
Instead the proposal is to implement this in the CMSIS Solution extension. The MDK uVision PackInstaller maintains the file ./Web/update.cfg containing the date of the last update. On initialization of the CMSIS Solution extension the file would be read. If the date is not the current date, then the cpackget update-index is executed and the current date writting to the update.cfg file.

ReinhardKeil, 2024-10-09:
We are aware of the issue and work on towards user-friendly solution. For now use this as a workaround.

Open the Terminal window and enter the command:
```
cpackget update-index
```

KimmoVaisanen-AlifSemi, 2024-10-09:
Ok. Thank you.

Do you happen to have any schedule for this yet?

ReinhardKeil, 2024-10-09:
Yes, it will be part of CMSIS-Toolbox 2.7 planned for December 2024.

KimmoVaisanen-AlifSemi, 2024-10-09:
Thank you again. 

Since you have a plan for this, this ticket can be closed. Unless you want to keep it for tracking purposes.

ReinhardKeil, 2024-10-18:
This will be solved by https://github.com/Open-CMSIS-Pack/cpackget/issues/355 + https://github.com/Open-CMSIS-Pack/cpackget/issues/358

closed 2024-10-18 by jkrech