ARM-software/vscode-cmsis-csolution #77: Packs that are downloaded require restart of VSCode to become available in Software Components view

ReinhardKeil, 2024-10-11:
Packs that are downloaded using `cbuild` with `--pack` option require restart of VSCode to become available in Software Components view.

**Steps to Reproduce**

Install on a new computer and download a BSP, i.e. from https://www.keil.arm.com/packs/ensemble-alifsemiconductor/boards/ using cpackget.  

New packs are getting downloaded, i.e. CMSIS 5.9 (as the example has a hard dependency). But the Software Components view does not show new components (as there is no pack reload triggered in the IDE).  It requires a restart of VS Code to see the new packs.

**Expected behavior**
A download of new packs should trigger a pack reload in the IDE (similar to uVision).

**Environment:**
- OS and Version: MacOS
- Extension Version: 1.40
- Target Device: Alif

jkrech, 2024-10-11:
Agreed. Instead of restarting VSCode you could use the "Refresh" from the CMSIS View `...` menu. This is triggering a reload of pdsc files in the cmsis-core-tools however only if a csolution project is currently loaded.

jkrech, 2024-11-11:
In CMSIS-Solution V 1.42 the "Refresh" button has been extended to  "Refresh (reload packs, update RTE).
There is no file watch on `$CMSIS_PACK_ROOT/pack.idx` triggering CMSIS-Core-Tools to reread pdsc files of installed packs.

jkrech, 2024-11-25:
pack.idx shall be watched by the CMSIS-Solution Extension as part of the CMSIS-Toolbox queue to reduce the amount of triggers until after `cbuild setup` completed.

closed 2024-12-09 by edriouk

JonatanAntoni, 2024-12-10:
cbuild setup is now triggered multiple times when using cpackget on another terminal. It looks like cpackget touches the pack.idx file multiple times or we are getting too many change events.

edriouk, 2024-12-10:
Watch is moved  to toolbox manager to sync with its queue. This triggers only reload of the packs, not another cbuild  setup run