ARM-software/vscode-cmsis-csolution #78: Add "Update RTE config files" to IDE

ReinhardKeil, 2024-10-11:
In v1.40 the behavior of the IDE changed, as `cbuild` with option `--update-rte` is only used when *csolution project files* are modified. This change ensures that files are modified when a project is just viewed (and IMHO correct behavior).

Unfortunately many examples do not ship with the RTE directory content as recommended under [reproduceable builds](https://github.com/Open-CMSIS-Pack/cmsis-toolbox/blob/main/docs/build-overview.md#reproducible-builds). The consequence is that the tool reports that the option `--update-rte` should be used, but in the IDE there is no easy way to apply it.

**Proposal**
Add a command "Update RTE config files" to this menu:
[image]

It should call `cbuild setup` with option `--update-rte` in the same way as when project files are modified.

jkrech, 2024-10-14:
Note: "Refresh" probably should be named "Reload", but this command is a manual trigger for:
a) cmsis-core-tools re-read of pdsc files - e.g. after installing new packs
b) `cbuild setup ... --context-set --update-rte [--packs]`
Is it worth having a separate command to just run b) independently?

ReinhardKeil, 2024-10-14:
I like that.  We could name it "Reload packs and update RTE"

jkrech, 2024-11-11:
In CMSIS Solution v 1.42 "Refresh" has been extended to "Refresh (reload packs, update RTE)