ARM-software/vscode-cmsis-csolution #82: Improve "Run and Debug" Configuration

ReinhardKeil, 2024-10-12:
The "Run and Debug" Configuration of the CMSIS Solution is very flexible, but needs review.

- Documentation: How are 3rd Party debuggers integrated, how are settings stored
- Documentation: How are "launch.json" dialogs created (see 2nd picture)
- Potential simplification: it manages multiple targets but this is not obvious
  - Should "Solution" change to "Target Type" and instead each target configuration accessible.
  - How are configurations for projects managed; is this a required feature?
- Review of parameters
  - Flash Algorithms come from DFP and BSP; however just one cmsisPack seems to be possible.
  - Can defaults be changed (i.e. command:arm-debugger.getApplicationFile)
 
[image]

This dialog opens on edit - it is a graphic view to launch.json
[image]

jreineckearm, 2024-10-12:
@ReinhardKeil , some initial input

> Documentation: How are "launch.json" dialogs created (see 2nd picture)

The dialogs are custom webviews provided by a [custom text editor](https://code.visualstudio.com/api/extension-guides/custom-editors#custom-text-editor) in the debug adapter extension. In this case living in the Arm Debugger extension. These allow to either show the text of the VS Code `launch.json`/`tasks.json`, or the webview overlay.

> Review of parameters

Most of this is handled by the debug adapter extension. In this case the Arm Debugger extension. The debug launch/attach configuration definitions and default configuration providers live there. Hence, changes need to be applied in the Arm Debugger extension.

> Flash Algorithms come from DFP and BSP; however just one cmsisPack seems to be possible.

This is a current limitation of the Arm Debugger extension. At the moment `cmsisPack` defines which pack to extract the required CMSIS device/debug descriptions from. This needs to change in the Arm Debugger extension for the above scenario.

> Can defaults be changed (i.e. command:arm-debugger.getApplicationFile)

They can by changing the default configuration providers (https://code.visualstudio.com/api/references/vscode-api#DebugConfigurationProvider and https://code.visualstudio.com/api/references/vscode-api#TaskProvider&lt;T&gt;) which are implemented in the debug adapter extension that contributes the debug configuration type.

In the Arm Debugger extension case, the currently used `arm-debugger.getApplicationFile` command tries under the hood a couple of known commands from other Arm extensions if present (`cmsis-csolution.getBinaryFiles`, `cmsis-csolution.getBinaryFile`) before it scans the folder structure for the first found application file (file endings defined in the Arm Debugger extension settings).
[image]

ReinhardKeil, 2025-05-12:
Done from my side.

closed 2025-05-12 by ReinhardKeil