ARM-software/vscode-cmsis-csolution #88: Extension version 1.40.0 is incompatible with cmsis-toolbox 2.4.0

KimmoVaisanen-AlifSemi, 2024-10-15:
Type: Bug Report

**Describe the bug**
- OS and Version: Ubuntu 22.04
- VS Code Version: 1.94.2
- Extension Version: 1.40.0
- Target Device:
- Other extensions you installed (and if the issue persists after disabling them):
- A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Clone https://github.com/alifsemi/alif_vscode-template
2. install submodules (`git submodule update --init`)
3. Open project with VSCode
4. Run Task `First time pack installation`
5. Build the project from CMSIS menu

**Expected behavior**
Applications build successfully

**Code sample and logs**
Build fails for missing xxxx.cbuild-set.yml file.

> *  Executing task: cmsis-csolution.build: Build 
>
> cbuild /home/.../alif.csolution.yaml --context-set
> info cbuild: Build Invocation 2.4.0 (C) 2024 Arm Ltd. and Contributors
> /home/.../alif.cbuild-pack.yml - info csolution: file generated successfully
> warning csolution: unable to locate /home/.../alif.cbuild-set.yml file.
> /home/.../alif.cbuild-idx.yml - info csolution: file generated successfully
> ...
> error cbuild: open /home/.../alif.cbuild-set.yml: no such file or directory

**Additional context**
Project builds with cmsis-toolbox version 2.6.0, but we have seen a lot of different issues with that as well. Also we cannot use toolbox newer than 2.4.0 with older Ensemble DFP packs due to "same files in multiple components problem" in our DFP pack. This is fixed in Ensemble DFP 1.3.0 but some users might want to use older packs.

jkrech, 2024-10-15:
Thanks for the feedback. You are correct there is a dependency of the updated CMSIS Solution Extension 1.40 on the CMSIS-Toolbox 2.6.1. 
Please note that the "same files in multiple components problem" gets worked around in 2.6.1 and therefore your older DFPs should work. Please let us know if that is not the case.

KimmoVaisanen-AlifSemi, 2024-10-15:
Ok, thank for the information. I tried 2.6.1 with the project we had problems with and it seems to build ok with our older DFP packs.

I close this ticket.

closed 2024-10-15 by KimmoVaisanen-AlifSemi