ARM-software/vscode-cmsis-csolution #91: Simplify access to Linker Map Files

ReinhardKeil, 2024-10-17:
As shown here:

[image]

This raises two questions:
 - is it possible to show a different tooltip text for "Open File" depending on the item where it is used?
 - do we need additional information to build information files or could this be created dynamically depending on the existence of the `*.MAP` file in the `out` directory.

mguzmanm, 2025-04-04:
> is it possible to show a different tooltip text for "Open File" depending on the item where it is used?

Yes, it is possible.

mguzmanm, 2025-04-04:
> do we need additional information to build information files or could this be created dynamically depending on the existence of the *.MAP file in the out directory.

No, we don't. @edriouk: thanks for clarification.

ReinhardKeil, 2025-04-04:
Ok, as this is possible, I suggest to show

On projects "Open cproject.yml file" instead of "Open File"

[image]

On layers "Open clayer.yml file" instead of "Open File"

[image]

mguzmanm, 2025-04-22:
Required changes were merged in main branch.

mguzmanm, 2025-04-23:
Reviewing 'linker' group's decoration issue.

Issue: the node "linker" is grey rather than white when a map file exists.

mguzmanm, 2025-04-23:
[image]

mguzmanm, 2025-04-23:
This behaviour appears when setting property 'resourcePath' to the tree item 'linker'.

`node.setAttribute('resourcePath', mapFilePath);`

The root cause analysis is still ongoing.

mguzmanm, 2025-04-23:
Notice that 'project' node also has a different decoration comparing with other groups.

[image]

mguzmanm, 2025-04-25:
In conclusion, the grey color comes from using Git, as the node refers to a file that is listed in .gitignore. If the workspace is not a Git repository, the 'linker' node will not be decorated.

Thanks @jkrech for bringing this up.

mguzmanm, 2025-05-09:
Changes were already merged in main branch (Commit [c9fbdf5](https://github.com/Arm-Debug/vscode-cmsis-csolution/commit/c9fbdf5348601adc8877fb19eb4eb91f4e873aa6))

mguzmanm, 2025-05-13:
Required changes after code review were performed.

closed 2025-05-13 by mguzmanm