ARM-software/vscode-cmsis-csolution #92: Cannot change context set when *csolution project* contains errors

ReinhardKeil, 2024-10-17:
When a *csolution project* contains errors (i.e. incorrect options), the Context Set dialog does not accept changes.

[image]

Work-around:
  - Ensure that there are no errors in the *csolution project*
  - or, edit the `cbuild-set.yml` file manually.

Note: There is work in progress which changes the way how context set is modified and how convert errors are shown in the IDE. After completion of this work, this ticket should be reviewed

huiwangbcm, 2024-10-21:
This issue appears on version 1.40.0. The cbuild-set.yml will not get updated with context set option. It also failed to generate new cbuild-set.yml if the file does not exit initially.

jkrech, 2024-10-22:
Correct, the issue is reported against version 1.40.0 of the Arm CMSIS Solution extension.
Could you share your project or what output you find in "Output" Tab when you selection "CMSIS Build Manager" from the drop down for the channel?
a) If there is no `*.cbuild-set.yml` file present, the invocation of `cbuild setup` will create one picking the first listed target-type and build-type with the first project. This does not seem to work for you which could mean that either you do not have a CMSIS-Toolbox bin directory in your path. (do you use the Arm Environment Manager extension and do you have a vcpkg-configuration.json file in your workspace?)
or the csolution/cproject.yml files are significantly broken such that the `cbuild setup` invocation exits without creating a cbuild-set.yml file.

huiwangbcm, 2024-10-22:
> Correct, the issue is reported against version 1.40.0 of the Arm CMSIS Solution extension. Could you share your project or what output you find in "Output" Tab when you selection "CMSIS Build [...]

I have purposed removed the old cbuild-set.yml. After executing a build, nothing appeared under the output (CMIS Build Manager). Here's the error message from the Terminal Tab,

[image]

Here's my typical build task in tasks.json,

{
	"type": "cmsis-csolution.build",
	"solution": "${workspaceRoot}/**/**/build/*csolution.yml",
	"useContextSet": false,
	"contexts": [
		"AppFW+AppFW"
	],
	"schemaCheck": false,
	"problemMatcher": [],
	"label": "cmsis-csolution.build: BuildApp"
},

jkrech, 2024-10-22:
It looks like you are still using CMSIS-Toolbox 2.3.0.
Could you please upgrade to the latest CMSIS-Toolbox 2.6.1.
The extension relies on the "cbuild setup" command being executed for the currently configured `*.cbuild-set.yml`. If that file does not exist a default selection will be generated and then used. E.g. the CMSIS outline view of the project used the cbuild-idx and cbuild YAML files for displaying detailed information. 
This process is invoked, whenever a solution is loaded or a csolution,cproject,clayer.yml files is updated. You can manually trigger it using the "CMSIS: Refresh" command.

closed 2024-11-04 by ReinhardKeil