ARM-software/vscode-cmsis-csolution #93: Change management dialog for "Run and Debug Configuration"

ReinhardKeil, 2024-10-17:
The tool supports already Run and Debug Configurations that are specific to a target-type.  However this feature is not very obvious.

What is not working yet, is the project specific configuration.  It is also unclear if this feature is really required (currently we cannot see any use case for that).

**Suggestion**

Replace the "Solution" list that currently shows the various projects with a "Target Type" list that shows Run and Debug Configuration for the different target types.  In the example below, the target-type with "-RAM" executes from RAM and does not require Flash download, whereas the other target-type requires Flash download.  This different configurations can be easily managed.

[image]

jkrech, 2024-12-06:
Renaming from "Solution" to "Target Type" will be in version 1.46.

ReinhardKeil, 2025-05-12:
Covered now in https://github.com/ARM-software/vscode-cmsis-csolution/issues/235

closed 2025-05-12 by ReinhardKeil

ReinhardKeil, 2025-05-22:
Replaced by [#235](https://github.com/ARM-software/vscode-cmsis-csolution/issues/235)