ARM-software/vscode-cmsis-csolution #94: Templates show now description line

ReinhardKeil, 2024-10-18:
**Describe the bug**

Examples, Templates, and Layers have a brief description.  However for Templates the description is not shown.

[image]

**Expected behavior**

Show the description that is in the PDSC file:
https://github.com/ReinhardKeil/STM32U5xx_DFP/blob/main/Keil.STM32U5xx_DFP.pdsc#L1923

RobertRostohar, 2024-10-24:
Would be also good to display the origin of the templates and also origin of the reference applications and examples. Maybe displayed when hovering over. If it comes from a certain pack, the pack version could be shown, or if it comes from the web.

thorstendb-ARM, 2024-12-04:
Description is shown in the DataManager "generalized" Version of the Tree Dropdowns:
[image]

What description should we put on the hard-coded, internal ones?

ReinhardKeil, 2024-12-04:
@thorstendb-ARM  

Brief description for internal Templates:

Blank: Start from scratch with a `main.c` and CMSIS device startup
TrustZone: Start from scratch with secure/non-secure project

thorstendb-ARM, 2024-12-04:
Done:
[image]

jkrech, 2024-12-06:
Confirmed with internal build from main branch: 2024-12-05

closed 2024-12-06 by jkrech