ARM-software/vscode-cmsis-csolution #96: Templates with multiple targets

ReinhardKeil, 2024-10-21:
The pack https://github.com/VladimirUmek/NXP_FRDM-MCXN947_BSP/releases contains a template with two targets: "Device

https://github.com/VladimirUmek/NXP_FRDM-MCXN947_BSP/releases

ReinhardKeil, 2024-12-05:
maybe we should restructure it and move it to build-type.

thorstendb-ARM, 2025-01-20:
Question: The template currently specifies also variables, device, board(?).
I guess only variables should be copied over and device/board should be overwritten?

thorstendb-ARM, 2025-01-20:
See https://github.com/Arm-Debug/vscode-cmsis-csolution/pull/2067

Todo:
- Decide which char to use as an indicator, as % is reserved in YAML
- Currently targetTypes from Template and CSolution are multiplied, reduce to one possible targetType from Solution only

Elise-Kaminski, 2025-02-10:
As discussed with Joachim Krech recently, the extension will handle this transparently without any user interaction.

jkrech, 2025-02-17:
@thorstendb-ARM, testing an internal build today, I found that `${Name}` is the "variable" that is replaced using the `target name`
`device:` and `board:` nodes are replaced by selected user selection as expected.

thorstendb-ARM, 2025-02-24:
@ReinhardKeil , is the behavior as expected?

ReinhardKeil, 2025-02-24:
yes works as expected.  But need to add this to documentation

jkrech, 2025-02-27:
Raised this issue for documenting in CMSIS-Toolbox:
https://github.com/Open-CMSIS-Pack/devtools/issues/1998

closed 2025-02-27 by jkrech