ARM-software/vscode-cmsis-csolution #99: Create new solution is broken for simple examples

RobertRostohar, 2024-10-24:
Type: Bug Report

**Describe the bug**
- OS and Version: Windows 11
- VS Code Version: 1.94.2
- Extension Version: 1.40
- Target Device: STM32

Starting a new solution based on existing CMSIS Solution examples incorrectly copies the contents from the pack. It seems that the project directory is copied twice and not completely. Sometimes only some files are copied and sometimes some files are corrupted (have only partial contents). 

Also note that the specified 'Solution Name' is only used for the directory name within the 'Solution Location'. The actual name of the solution is taken from the pack. This is misleading.

**To Reproduce**
Steps to reproduce the behavior:

Install at least the following CMSIS packs:
- Keil::STM32H7xx_DFP@3.0.0
- Keil::STM32H743I-EVAL_BSP@1.0.0
- Keil::STM32H745I-DISCO_BSP@1.0.0

Example A: Simple example (solution and project yml files in the same directory)
1. Go to 'Create a New solution'
2. Select 'Target Board': STM32H743-EVAL
3. Select 'Template, Reference Applications, and Examples:' CMSIS SOLUTION EXAMPLE - Blinky
4. Chose 'Solution Location'
5. Disable 'Initialize Git repository'
6. Click on 'Create'
7. Go to 'Explorer' and observe the copied files to the Solution location. It will contain duplicates in the 'Blinky' sub-directory. But some files are missing (ex: 'retarget_stdio.c'), and build fails.

[image]

Example B: Simple example (solution with two projects in sub-directories)

Steps are the same as for example A, except that the 'Target Board' is STM32H745I-DISCO.
Results also in duplicated files when copying the project and with missing or corrupted files.

**Expected behavior**
The project files from the pack should be copied to the specified location as they are found in the pack (no duplication or corruption).

ReinhardKeil, 2024-10-24:
Internal reference MCUTO-380

thorstendb-ARM, 2024-12-04:
Seems the web examples (zip) are broken. See link (download & unzip), it contains itself in the exaple. Hardto reproduce when not knowing the source of the project, as the assumption, both (local and online) could be the same, is wrong.
https://www.keil.arm.com/boards/stmicroelectronics-stm32h745i-disco-revb-6f78fb3/projects/

ReinhardKeil, 2024-12-04:
For the 745, the content in the pack is
https://github.com/Open-CMSIS-Pack/STM32H745I-DISCO_BSP/tree/main/Examples/Blinky

Whereas the downloaded ZIP file seems to contain the csolution project twice:
[blinky-229ef2dd.zip](https://github.com/user-attachments/files/18008410/blinky-229ef2dd.zip)

jkrech, 2025-02-26:
Not fixed in the timelines for 1.50 -> hence rescheduled to 1.52

JonatanAntoni, 2025-03-25:
The example archive [blinky-229ef2dd.zip](https://github.com/user-attachments/files/18008410/blinky-229ef2dd.zip) is still not fixed.

jkrech, 2025-07-14:
Pack Keil::STM32H745I-DISCO_BSP@1.1.0 was released and the example was updated

Downloading the example now:
https://downloads.software.api.keil.arm.com/projects/blinky-229ef2dd.zip
does not show any unexpected duplication issues anymore.

closed 2025-07-14 by jkrech