ARM-software/vscode-cmsis-csolution #100: Fault example from pack Arm:CMSIS-View@1.2.0 not shown in Create New Solution with Rev.C board

ReinhardKeil, 2024-10-25:
Version: VS Code CMSIS Solution 1.40

I would expect to see the "Fault" example.  It is not shown.

[image]

jkrech, 2024-10-25:
After adding the `revision=` attribute to the example in the pack description file, the example shows for me.
Is the general expectation for an example that specifies a board without the `revision` would show for boards with any `revision` or no `revision` specified at all.

Currently the only work around I see is editing the installed pdsc file.

ReinhardKeil, 2024-10-25:
When no `revision` is specified in the example it should show up for any board revision.

JonatanAntoni, 2024-11-12:
This is expected to work in latest main:
[image]

It may take a while to load the examples. The dropdown shall be updated after a view seconds to list "LOCAL CMSIS SOLUTION EXAMPLES".

closed 2024-11-12 by jkrech