ARM-software/vscode-cmsis-csolution #101: Unable to flash a device with arm-debugger

MiloradCvjetkovic, 2024-10-29:
Type: Bug Report

A simple **Blinky** from the BSP cannot be **flashed** to the device or debugged with **arm-debugger**.

**Expected behavior**

Be able to download the executable image to the device's flash memory.

**Environment**

- OS and Version: Windows 11 Enterprise, 23H2
- VS Code Version: 1.94.2
- Extension Version: **1.40.0**
- Target Device: **STM32H743XIHx**

**To Reproduce**

Install the following packs (with clean/empty packs folder) in the order specified, manually (not with cpackget):
1. [ARM.CMSIS.6.1.0.pack](https://www.keil.com/pack/ARM.CMSIS.6.1.0.pack)
2. [ARM.CMSIS-Compiler.2.1.0.pack](https://www.keil.com/pack/ARM.CMSIS-Compiler.2.1.0.pack)
3. [ARM.CMSIS-Driver_STM32.1.0.0.pack](https://www.keil.com/pack/ARM.CMSIS-Driver_STM32.1.0.0.pack)
4. [ARM.CMSIS-RTX.5.9.0.pack](https://www.keil.com/pack/ARM.CMSIS-RTX.5.9.0.pack)
5. [ARM.CMSIS-View.1.2.0.pack](https://www.keil.com/pack/ARM.CMSIS-View.1.2.0.pack)
6. [Keil.STM32H7xx_DFP.4.0.0.pack](https://www.keil.com/pack/Keil.STM32H7xx_DFP.4.0.0.pack)
7. [Keil.STM32H743I-EVAL_BSP.1.0.0.pack](https://www.keil.com/pack/Keil.STM32H743I-EVAL_BSP.1.0.0.pack)

- Connect the **STM32H743I-EVAL** board via **ST-LINK/V2_1** USB port to a USB port on the PC
- Copy the **Blinky** from installed STM32H743I-EVAL_BSP somewhere on the disk
- Open **Blinky.csolution.yml** with VS Code
- Accept **Add Software Layer** - **Select Compiler** - **AC6** and click on **OK** button
- Click on **CMSIS** extension and select **Rebuild**
- Click on **Run** button, select **Add run task...**, select **arm-debugger.flash: Flash Device**, save the generated **tasks.json**
- Click on **Run** button, select **arm-debugger.flash: Flash Device**, select **STM32 STLink**
- Following message should pop-up
  [image]

> Note: this bug cannot be consistently reproduced on all PCs, however, on my PC it is consistently reproducible.

**Workaround**

Changing the following entry in the **tasks.json** :
```
"cmsisPack": "${command:cmsis-csolution.getTargetPack}",
```
to 
```
"cmsisPack": "${command:device-manager.getDevicePack}",
 ```
fixes the problem, and device can be successfully flashed.

jkrech, 2024-11-04:
I was able to replicate the problem.
There seems to be a glitch in `CMSIS-Core-Tools` failing to provide the DFP of the "used device" when using the feature command:
```
cmsis-csolution.getTargetPack
```
For me temporarily commenting the `device:` or `board:` node avoided the error.

MiloradCvjetkovic, 2024-11-05:
This is a simple setup with just a handful of packs and commenting might help but I have seen sometimes in more complex situations that even commenting does not help.

jkrech, 2024-11-08:
Could you please try to reproduce this problem again and then run the "Refresh" from the `...` menu of the CMSIS Solution Extension and then try it again. Note that for testing you can also define a task in task.json:

```
        {
            "label": "TEST cmsis-csolution commands",
            "type": "shell",
            "command": "echo 'DEVICE: ${command:cmsis-csolution.getDeviceName} \nPACK: ${command:cmsis-csolution.getTargetPack}'",
            "problemMatcher": []
         }
```
When you copy/unzip pack versions manually, please ensure that you follow the rule:
$CMSIS_PACK_ROOT/<vendor>/<packname>/<version>/

MiloradCvjetkovic, 2024-11-11:
The `...` and `Refresh` did not change anything, still fails to flash with same error.

That information for debugging of `tasks.json` is useful, however, the result with using suggested `command` is as below:

with:
```"command": "echo 'DEVICE: ${command:cmsis-csolution.getDeviceName} \nPACK: ${command:cmsis-csolution.getTargetPack}'"```
**nothing gets output in the Terminal**

but with:
```"command": "echo 'DEVICE: ${command:device-manager.getDeviceName} \nPACK: ${command:device-manager.getDevicePack}'"```
result in the Terminal is:
```
DEVICE: STM32H743XIHx
PACK: Keil::STM32H7xx_DFP@4.0.0
```

JonatanAntoni, 2024-11-11:
@MiloradCvjetkovic, could you please check if a "Refresh solution" solves your issue? On the CMSIS pane look for the tree-dots-menu.

MiloradCvjetkovic, 2024-11-11:
As already written @JonatanAntoni , it did not solve the problem.

BTW, there is no `Refresh solution` in the `...`, there are `Rebuild solution` and `Refresh`, and that `Refresh` did not help.

jkrech, 2024-11-11:
A fully initialized $CMSIS_PACK_ROOT directory is a pre-requisite. Once $CMSIS_PACK_ROOT/.Web/index.pidx is located by cmsis-core-tools, the command works as expected.

MiloradCvjetkovic, 2024-11-12:
That was an attempt to provide a simple reproducible failure to flash the device.

However, in my more complicated environment with multiple DFPs/BSPs and existing `.Web/index.pidx` , executing just :
`"command": "echo 'PACK: ${command:cmsis-csolution.getTargetPack}'"`
I still get the same :
[image]

jkrech, 2024-11-12:
Another user reported problems with this example.
https://www.keil.arm.com/boards/nxp-frdm-k32l3a6-989d2e5/projects/

edriouk, 2024-11-28:
I have managed getDeviceName and getTargetPack commands to read values from cbuild.yml file.
getProcessorName reurns the value only if case in a single context. Otherwise a empty string is returned. 
We need to discuss how to pass an argument with a context name or at least with a project name to the command. 
It should be feasible for a command triggered from a selected item in Outline View. However I do  not know how to pass an argument via tasks.json

edriouk, 2024-12-04:
This PR also solves the getProcessorName: https://github.com/Arm-Debug/vscode-cmsis-csolution/pull/1990

jkrech, 2024-12-06:
Got confirmation from a number of internal users, that this works on internal build from main.

closed 2024-12-06 by jkrech