ARM-software/vscode-cmsis-csolution #102: Print error message from Generator in Notification

HansSchneebauer, 2024-10-29:
It would be good to see the error messages of a generator in the corresponding [notification](https://code.visualstudio.com/api/ux-guidelines/notifications). 
In the screen capture below, I started the generator (here CubeMX), which opened a notification (see red rectangle). Since CubeMX was not installed on my machine it did not work. The notification 'Starting generator...' was shown there for a long time and no error message was visible. It would be good to see the error message (green rectangle) in the notification because when the Output window of the 'CMSIS Build Manager' is not open by chance, a user does not see an error. This request is also related to [request #98](https://github.com/ARM-software/vscode-cmsis-csolution/issues/98).

[image]

JonatanAntoni, 2025-02-05:
Relates to #121

JonatanAntoni, 2025-02-05:
@HansSchneebauer, the actual error message is only available in the output log as it comes from a sub-process. What can easily be done is a generic error notification and putting the output window into focus. Please check Arm-Debug/vscode-cmsis-csolution#2111 if this is sufficient for you.

closed 2025-02-11 by JonatanAntoni

jkrech, 2025-02-17:
The actual error message is only visible in "output" channel for "CMSIS Solution". The pop up message reads:

[image]