ARM-software/vscode-cmsis-csolution #103: Project Lifecycle Management (PLM) of Config Files - Upgrade Config File

ReinhardKeil, 2024-10-30:
A powerful feature of the CMSIS-Toolbox is [PLM of configuration file](https://github.com/Open-CMSIS-Pack/cmsis-toolbox/blob/main/docs/build-overview.md#plm-of-configuration-files)

This is a potential workflow for the upgrade of configuration files in VS Code where:
 - A `new config file view` compares the current config file with a new version of the same config file. This allows to merge user settings to the new config file.
 - A `current config file view` compares the current config file with the original unmodified config file. This shows the user settings in project.

The mock-up below shows this and highlights components with config file updates in the CMSIS view:

[image]

The highlighted component has two additional **Action Buttons** that start the process:
 - `Upgrade Config File` - [potential icon](https://github.com/microsoft/vscode-codicons/blob/main/src/icons/git-pull-request-create.svg) - only shown when `.update@version` file present
- `Confirm Config File Upgrade` - only shown when `.new@version` file present

> Note: in case that a component has more then one config file to upgrade a list "Select Config File" is shown.

**Action: Upgrade Config File**
- If `.new@version` file not exist:
    - copy `.update@version` file to `.new@version` file and remove R/O attribute
    - create backup: copy original file to `original file.bak`
- Start 'compare view' with original file (`RTX_Config.c`) and `.new@version` file
- Start 'compare view' with original file and `.base@version` file

**Action: Confirm Config File**
- Compare original file with `orignal file.bak`
   - if different stop and output: "Only the new configuration file should be changed; changes to original file detected"
- Rename `.base@version` file to `.base.bak@version`
- Rename `.update@version` file to `.base@version`
- Delete original file
- Rename `.new@version` file to orignal file.

jkrech, 2024-11-05:
VSCode also offers a ["three way" merge editor](https://code.visualstudio.com/docs/sourcecontrol/overview#_3way-merge-editor) which can be launched for git merge conflicts.
[image]

brondani, 2024-12-05:
At least in the command line the 3 way merge can be freely invoked:
```
$ code -h
...
  -m --merge <path1> <path2> <base> <result> Perform a three-way merge by
                                             providing paths for two modified
                                             versions of a file, the common
                                             origin of both modified versions
                                             and the output file to save merge
                                             results.
```

For example:
```
cp RTX_Config.h RTX_Config.h.bak
code --merge RTX_Config.h.bak RTX_Config.h.update\@5.6.0 RTX_Config.h.base\@5.5.2 RTX_Config.h
```
[image]

There are many features that are maybe not so evident. Explore the buttons: for example clicking on "Compare With Base" opens another view:
[image]

mguzmanm, 2025-05-06:
A quick implementation shows that it's possible to call 3 way merge editor from the Outline view without opening a new instance of vscode. 

[image]

ReinhardKeil, 2025-07-29:
Please change messages in the popup window to:
?   'RTX_Config.c' has corrections. A file update is suggested.
!    'RTX_Config.c' has new features. A file update is recommended.
X   'RTX_Config.c' is incompatible. A file update is required.

Instead of adding (?), (!), (X) to the file/component, the symbol ?, !, X could be added at the end similar to 'U' or 'P' with hover over messages and color indication.  Suggested hover over messages

? file update suggested
! file update recommended
X file update required

closed 2025-07-31 by jkrech