ARM-software/vscode-cmsis-csolution #105: Improve startup time, just one execution of `cbuild setup` is required

ReinhardKeil, 2024-10-31:
When starting VS Code, two times the command `cbuild setup` is issued.

Enqueue: cbuild setup c:\Test\Oct14\USBD\USB_Device.csolution.yml --context-set
Execute: cbuild setup c:\Test\Oct14\USBD\USB_Device.csolution.yml --context-set
Enqueue: cbuild setup c:\Test\Oct14\USBD\USB_Device.csolution.yml --context-set
+--------------------------------------------------
(1/3) Setting up context: "HID.Debug+STM32U585AIIx"
[1/1] C:\WINDOWS\system32\cmd.exe /C "cd /D C:\Test\Oct14\USBD\tmp\1 && "C:\Program Files\CMake\bin\cmake.exe" -E make_directory C:/Test/Oct14/USBD/out/HID/STM32U585AIIx/Debug && "C:\Program Files\CMake\bin\cmake.exe" -E copy_if_different C:/Test/Oct14/USBD/tmp/1/compile_commands.json C:/Test/Oct14/USBD/out/HID/STM32U585AIIx/Debug"

+----------------------------------------------------------
(2/3) Setting up context: "MassStorage.Debug+STM32U585AIIx"
[1/1] C:\WINDOWS\system32\cmd.exe /C "cd /D C:\Test\Oct14\USBD\tmp\2 && "C:\Program Files\CMake\bin\cmake.exe" -E make_directory C:/Test/Oct14/USBD/out/MassStorage/STM32U585AIIx/Debug && "C:\Program Files\CMake\bin\cmake.exe" -E copy_if_different C:/Test/Oct14/USBD/tmp/2/compile_commands.json C:/Test/Oct14/USBD/out/MassStorage/STM32U585AIIx/Debug"

+---------------------------------------------------------
(3/3) Setting up context: "VirtualCOM.Debug+STM32U585AIIx"
[1/1] C:\WINDOWS\system32\cmd.exe /C "cd /D C:\Test\Oct14\USBD\tmp\3 && "C:\Program Files\CMake\bin\cmake.exe" -E make_directory C:/Test/Oct14/USBD/out/VirtualCOM/STM32U585AIIx/Debug && "C:\Program Files\CMake\bin\cmake.exe" -E copy_if_different C:/Test/Oct14/USBD/tmp/3/compile_commands.json C:/Test/Oct14/USBD/out/VirtualCOM/STM32U585AIIx/Debug"

Execute: cbuild setup c:\Test\Oct14\USBD\USB_Device.csolution.yml --context-set
+--------------------------------------------------
(1/3) Setting up context: "HID.Debug+STM32U585AIIx"
[1/1] C:\WINDOWS\system32\cmd.exe /C "cd /D C:\Test\Oct14\USBD\tmp\1 && C:\Users\reikei01\.vcpkg\artifacts\2139c4c6\tools.kitware.cmake\3.28.4\bin\cmake.exe -E make_directory C:/Test/Oct14/USBD/out/HID/STM32U585AIIx/Debug && C:\Users\reikei01\.vcpkg\artifacts\2139c4c6\tools.kitware.cmake\3.28.4\bin\cmake.exe -E copy_if_different C:/Test/Oct14/USBD/tmp/1/compile_commands.json C:/Test/Oct14/USBD/out/HID/STM32U585AIIx/Debug"

+----------------------------------------------------------
(2/3) Setting up context: "MassStorage.Debug+STM32U585AIIx"
[1/1] C:\WINDOWS\system32\cmd.exe /C "cd /D C:\Test\Oct14\USBD\tmp\2 && C:\Users\reikei01\.vcpkg\artifacts\2139c4c6\tools.kitware.cmake\3.28.4\bin\cmake.exe -E make_directory C:/Test/Oct14/USBD/out/MassStorage/STM32U585AIIx/Debug && C:\Users\reikei01\.vcpkg\artifacts\2139c4c6\tools.kitware.cmake\3.28.4\bin\cmake.exe -E copy_if_different C:/Test/Oct14/USBD/tmp/2/compile_commands.json C:/Test/Oct14/USBD/out/MassStorage/STM32U585AIIx/Debug"

+---------------------------------------------------------
(3/3) Setting up context: "VirtualCOM.Debug+STM32U585AIIx"
[1/1] C:\WINDOWS\system32\cmd.exe /C "cd /D C:\Test\Oct14\USBD\tmp\3 && C:\Users\reikei01\.vcpkg\artifacts\2139c4c6\tools.kitware.cmake\3.28.4\bin\cmake.exe -E make_directory C:/Test/Oct14/USBD/out/VirtualCOM/STM32U585AIIx/Debug && C:\Users\reikei01\.vcpkg\artifacts\2139c4c6\tools.kitware.cmake\3.28.4\bin\cmake.exe -E copy_if_different C:/Test/Oct14/USBD/tmp/3/compile_commands.json C:/Test/Oct14/USBD/out/VirtualCOM/STM32U585AIIx/Debug"

jkrech, 2024-10-31:
We need to locate the trigger for the second invocation
"open csolution" - this will start watching the project input YAML files (`*.csolution`, `*.cproject`, `*.cgen`, `*.clayer` and `*.cbuild-set`)
Are we watching any other files?

brondani, 2024-11-04:
The second `cbuild setup` call is triggered by vcpkg activation.
Should we prevent this call? How to differentiate a vcpkg activation happening at startup and others happening at any other point in time - which I assume should not be suppressed?

jkrech, 2024-11-04:
Assuming that the user relies on the Arm Environment Manager, the first invocation of `cbuild setup` fails, because no CMSIS-Toolbox is in the PATH yet. 
On the other hand we don't know whether the user relies on the Arm Environment Manager and the current folder contains a `vcpkg-configuration.json` file and cannot wait for the Arm Environment Manager `activate` complete flag.
Am I missing the trick?

jkrech, 2024-11-11:
CMSIS-Solution Extension version 1.42 only invokes cbuild-setup once on loading the solution without "--update-rte".

closed 2024-11-11 by jkrech