ARM-software/vscode-cmsis-csolution #107: Improvements on CMSIS View: default collapse, number of components, add group to project, context menu, delete

ReinhardKeil, 2024-10-31:
These are suggestions to improve the Project Outline in the CMSIS-View

- By default, Components, linker, constructioned-files should be collapsed
  [image]

- Add on "Components" the number, i.e. "Components (10)" similar to the Manage Software Components view.

- Add at project level an Action Button to Add Groups

- Is it possible to add a context menu similar to Explorer View?

- Use 'Delete' button to remove groups and files.  There should be an dialog similar to the Explore View with the options 
   - "Remove from project only"
   - "Move to Recycle Bin"
   - "Cancel"

mguzmanm, 2024-12-06:
- [Done] By default, Components, linker, constructed files should be collapsed to put the focus on user code
- [Done] Add on "Components" the number, i.e. "Components (10)" similar to the Manage Software Components view.
- [Done] Add at project level an Action Button to Add Groups

jkrech, 2024-12-06:
Confirmed based on internal build from main branch 2024-12-05

closed 2024-12-06 by jkrech