ARM-software/vscode-cmsis-csolution #108: Add Templates: overwrites existing files, management for instances missing

ReinhardKeil, 2024-10-31:
The Add Templates feature provides access to example code that shows the usage of components.

[image]

However there are two problems:
- Existing files are overwritten.  Instead an error should pop-up that informs the user about the duplicate file name.  Note that a template can contribute multiple files, it is therefore important to check for existing files before coping any file to the user application.

- Templates can have instances (i.e. USB HID), but currently there is not way to enter the instance number.

mguzmanm, 2025-05-07:
Problem 2: 

> - Templates can have instances (i.e. USB HID), but currently there is no way to enter the instance number.

will be addressed in issue [#233](https://github.com/ARM-software/vscode-cmsis-csolution/issues/233)

mguzmanm, 2025-05-13:
Changes after code review were performed.