ARM-software/vscode-cmsis-csolution #109: Template Project with explicit Target "MyTarget" results incorrect project

ReinhardKeil, 2024-11-03:
When using Create New Project with explicit Target name as shown below
- Ambiq Micro::Apollo3 Blue EVB:Ver 1.0
- type: MyTarget

[image]

An incorrect project is created with:
- missing project
- that shows only "Loading" in the CMSIS View

[image]

**Expected behavior**
- Project is correctly generated
- CMSIS View tolerates such incorrect projects and shows an error.

ReinhardKeil, 2024-11-04:
Test project: [Blank_solution.zip](https://github.com/user-attachments/files/17620787/Blank_solution.zip)

jkrech, 2024-11-05:
@ReinhardKeil this is working correctly in the released version 1.40 but fails for the latest unpublished builds.
As far as I can see this is not related to the renamed target-type. The "Create Solution" dialog is missing the part that by default adds the `Project.cproject.yml ` project but also allows to add additional projects.

ReinhardKeil, 2024-11-12:
no, cannot duplicate it with the official release.

closed 2024-11-12 by ReinhardKeil