ARM-software/vscode-cmsis-csolution #111: Failed to resolve 'Device:Startup' component from 'legacy' ST DFP packs

chentang16, 2024-11-04:
Type: Bug Report

**Describe the bug**
- OS and Version: Windows 11
- VS Code Version: 1.95.1
- Extension Version: 1.40.0
- Target Device: STM32F303RET (NUCLEO-F303RE)

When creating a new STM32F303RET (NUCLEO-F303RE) csolution project from the latest STM32F3xx_DFP pack v2.2.2 (not clayer based), Software Components in VS Code failed to resolve 'Device:Startup' component, no matter if the "standalone" or the "STM32CUBEMX" variant is selected.

Build error output:
error csolution: no component was found with identifier 'Device:Startup'
error csolution: processing context 'test.Debug+STM32F303RETx' failed
Process failed with exit code 1
Process failed with exit code 1

The same new project creation process done in MDK v5 µVision IDE works fine as expected, as shown in the attached screenshot.
[image]

ReinhardKeil, 2024-11-04:
Could you add the csolution project as ZIP file?

chentang16, 2024-11-04:
Here is my csolution project
[Blank_solution.zip](https://github.com/user-attachments/files/17620006/Blank_solution.zip)

jkrech, 2024-11-04:
The selected "Template Project" is the "Blank Solution" which already has the component "Device:Startup" selected in the "Project.cproject.yml" file. 
However the STM32F3xx_DFP describes the "Device:Startup" component in two Cbundles "Standalone" and "STM32CubeMX" and because present Device:Startup has no bundle at all, the Component displays in the Variant column "None" , "Standalone", "STM32CubeMX"

You first need to unselect the "Device:Startup" component that cannot be resolved.
Afterwards you need to either select "Standalone" or "STM32CubeMX" from the variant column and select the "Device:Startup" component from the respective bundle.
Now you should see e.g. 
```
    - component: Keil::Device&Standalone:Startup
 
```
in the cproject.yml file"Device Startup"

ReinhardKeil, 2024-11-04:
We need to improve visiblity in the UX

chentang16, 2024-11-05:
> The selected "Template Project" is the "Blank Solution" which already has the component "Device:Startup" selected in the "Project.cproject.yml" file. However the STM32F3xx_DFP describes the "Device:Startup" component in two Cbundles "Standalone" and "STM32CubeMX" and because present Device:Startup has no bundle at all, the Component displays in the Variant column "None" , "Standalone", "STM32CubeMX"
> 
> You first need to unselect the "Device:Startup" component that cannot be resolved. Afterwards you need to either select "Standalone" or "STM32CubeMX" from the variant column and select the "Device:Startup" component from the respective bundle. Now you should see e.g.
> 
> ```
>     - component: Keil::Device&Standalone:Startup
> ```
> 
> in the cproject.yml file"Device Startup"
> 
> A problem of the component window is, that it allows to concurrently select components from different bundles without raising a warning/error. Also `cbuild` does not complain about two Device:Startup components:
> 
> ```
>   components:
>     - component: Keil::Device&Standalone:Startup
>     - component: Keil::Device&STM32CubeMX:Startup
> ```

Yes, thx! This indeed works. However, the first step "You first need to unselect the "Device:Startup" component that cannot be resolved" I have to repeat two/three times to properly unselect it. It seems to me each time of click reflects one validator error shown in red in my screenshot above.