ARM-software/vscode-cmsis-csolution #112: Build Message Window - extend size

ReinhardKeil, 2024-11-06:
Is it possible to make this window wider?

[image]

There is currently a line break which is not in the message itself.
```
Project.cproject.yml - dependency validation for context 'Project.Debug+STM32F301C6Tx' failed:
SELECTABLE Keil::Device&STM32CubeMX:STM32Cube Framework:STM32CubeMX@1.11.3
  require Device:STM32Cube HAL"
```

brondani, 2024-11-11:
This is a status bar `tooltip` property:
https://code.visualstudio.com/api/references/vscode-api#StatusBarItem

Unfortunately its max width is not exposed in the vscode API.

It seems the issue [Configure tooltip max width](https://github.com/microsoft/vscode/issues/14165) has been requested for years, eventually the editor's tootip has been made resizable but not persistent. The status bar tootip is not even resizable.

Anyway in my opinion errors/warnings should be displayed in the `PROBLEMS` panel rather than in the status bar tooltips.

ReinhardKeil, 2024-11-12:
Agree with you: we should use `Problems` to display this type of messages.  

I keep this open.