ARM-software/vscode-cmsis-csolution #113: CMSIS View does not show project

ReinhardKeil, 2024-11-06:
This project [adc_12b1msps_sar_interrupt.zip](https://github.com/user-attachments/files/17647380/adc_12b1msps_sar_interrupt.zip)
 compiles with no errors, but the project content is not shown in the CMSIS View

[image]

KeilChris, 2024-11-19:
I see a similar (the same?) issue with any project.
Steps to reproduce:

- open a solution
- go to the CMSIS view
- click on the "Manage Solution" cog wheel, and toggle the "Build Type" of the "Active Project".

Then the CMSIS view goes blank stating "Loading...". Only way to get it back is to use "Refresh (Reload Packs, Update RTE)".

ReinhardKeil, 2024-11-19:
I can confirm: the work-around use Refresh also works in my case.

ReinhardKeil, 2024-12-04:
Seems to be fixed in 1.47