ARM-software/vscode-cmsis-csolution #116: Do not show UV projects that require AC5 in Create new solution dialog

ReinhardKeil, 2024-11-07:
uVision examples that are created for AC5 cannot be converted. On the web side these examples are marked with AC5.

In the Create new solution dialog such examples should not be shown.

This is a board that has such examples https://www.keil.arm.com/boards/abov-starterkit-a31g112cl-v11-9cec7c0/projects/

JonatanAntoni, 2024-11-19:
Examples retrieved from Core Tools can't be filtered unfortunately. The web search results retrieved via Solar search contain a `convertible` flag indicating its compatibility (via uv2csolution). We plan to list only those uVision examples returned from Solar which are convertible.

See Arm-Debug/vscode-cmsis-csolution#1919

JonatanAntoni, 2024-11-27:
cc @thorstendb-ARM

jkrech, 2024-12-03:
I confirm that uVision projects from the Web that require AC5 are no longer displayed, whereas uVision projects from installed packs are listed even if they use AC5 as there is no information in the pdsc file indicating whether a project uses AC5 or AC6.

closed 2024-12-03 by jkrech