ARM-software/vscode-cmsis-csolution #117: New Solution workflow only shows Reference Examples for boards that have a board layer in an already installed pack

jkrech, 2024-11-08:
The "New Solution" workflow only lists "Reference Applications" if a "Board Layer" for the selected board is found (usually in the BSP).
Since the "BSP Pack" is only downloaded and installed by `cbuild setup` on a created project, the BSP must be downloaded manually or any other project being created first.
Ensure to run the "Refresh" before opening the "New Solution" workflow after the pack was installed.

This can only be solved, once the board details retrieved from the Web Service, also include information about available board layers.

jkrech, 2024-12-03:
This requires that layers are provided via the Web API which is currently not supported.

ReinhardKeil, 2025-02-04:
@jkrech is this already fixed?  If so, I suggest to close it as #147 covers the remaining problems.

JonatanAntoni, 2025-02-04:
@ReinhardKeil, this cannot be fixed without enhancements to the webservice. At the moment (v1.48.0) we only list reference examples if
- a board is selected
- any installed pack provides at least one board layer for that board, i.e., typically this requires the Board Support Pack to be already installed

ReinhardKeil, 2025-02-04:
ticket is parked for future enhancements - currently no action planned

closed 2025-02-05 by ReinhardKeil