ARM-software/vscode-cmsis-csolution #122: Show open document for API

ReinhardKeil, 2024-11-14:
The API documentation is not accessible when the component has own documentation.

Is it possible to add the API documentation to the API file instead and offer an Action icon?

[image]

jkrech, 2024-11-19:
Initial investigation suggests that should be doable and it has been added to the backlog for the upcoming release.

mguzmanm, 2024-11-27:
API documentation is now accesible as required.

closed 2024-11-27 by mguzmanm

jkrech, 2024-12-02:
Component documentation and API documentation are selectable (vsix from 2024-11-28)
[image]

Elise-Kaminski, 2024-12-04:
Documented with https://github.com/Arm-Debug/ks-vscode-extension-docs/pull/272