ARM-software/vscode-cmsis-csolution #126: `Create a New Solution` does not work correctly for examples from locally installed packs (`.csolution.yml` packs version problems)

MiloradCvjetkovic, 2024-11-19:
Type: Bug Report

**Extension Version**: 1.44.0

**Description**

Using `Create a New Solution` button for copying example behaves differently when example is coming from the Web or from locally installed packs.

If the example is copied from locally installed pack the pack description in the `.csolution.yml` is changed in the following ways:
- pack description specifying version using `^` is changed to `"::"`
- pack description specifying version using `>=` or `explicit version` looses the version description

> Note: If the example is coming from the Web the `.csolution.yml` file is unchanged.

**Example**

`Example.csolution.yml` with the following content:
```
  packs:
    - pack: Keil::STM32U5xx_DFP@^3.0.0
    - pack: Keil::B-U585I-IOT02A_BSP@>=2.0.0
    - pack: ARM::CMSIS-Driver_STM32@1.0.0
```

in the copied version looks like below:
```
  packs:
    - pack: "::"
    - pack: Keil::B-U585I-IOT02A_BSP
    - pack: ARM::CMSIS-Driver_STM32
```

**Expected behavior**

`.csolution.yml` should not be changed when copied from the locally installed pack.

The behavior is correct when example is coming from the Web, in that case the complete `.csolution.yml` file is preserved as is.

> Note: I could not test `^` in the Web example but I expect that it is retained as it looks like `.csolution.yml` is not touched when it comes from the Web

thorstendb-ARM, 2024-11-27:
"Local Examples" have been an extra data path in v1.44, ant too much code was copied over. Should be fixed in the next version.

thorstendb-ARM, 2024-12-03:
@MiloradCvjetkovic , please find vsix for testing here:
https://github.com/Arm-Debug/vscode-cmsis-csolution/actions/runs/12137393257

MiloradCvjetkovic, 2024-12-03:
Thanks @thorstendb-ARM , I have tested and it works well.

closed 2024-12-03 by MiloradCvjetkovic