ARM-software/vscode-cmsis-csolution #127: `Create a New Solution` does not work correctly for examples from locally installed packs (`.csolution.yml` target type problem)

MiloradCvjetkovic, 2024-11-19:
Type: Bug Report

**Extension Version**: 1.44.0

**Description**

Using `Create a New Solution` button for copying example behaves differently when example is coming from the Web or from locally installed packs.

If the example is copied from locally installed pack the `target type` in the `.csolution.yml` is set to name provided in the `Target Type` edit box in the `Create Solution` window.

> Note: If the example is coming from the Web the `.csolution.yml` file is unchanged.

This is a problem because in the examples that use Generator (like for B-U585) the generated files are related (their location) to the `target type` and if target type is changed then original generated files are not used (`cgen` file not found).

**Example**

`Example.csolution.yml` with the following content:
```
  target-types:
    - type: B-U585I-IOT02A
      board: STMicroelectronics::B-U585I-IOT02A
      device: STMicroelectronics::STM32U585AIIx
```

in the copied version looks like below (if the `Target Type` was left as default):
```
  target-types:
    - type: STM32U585AIIx
      board: STMicroelectronics::B-U585I-IOT02A:Rev.C
      device: STMicroelectronics::STM32U585AIIx
```

Resulting in following build error message:
```
Blinky.cgen.yml - error csolution: cgen file was not found, run generator 'CubeMX' for context 'Blinky.Debug+STM32U585AIIx'
error csolution: processing context 'Blinky.Debug+STM32U585AIIx' failed
```

**Expected behavior**

`.csolution.yml` should not be changed when copied from the locally installed pack.

The behavior is correct when example is coming from the Web, in that case the complete `.csolution.yml` file is preserved as is.

thorstendb-ARM, 2024-11-27:
Should be solved in the next version. "Local Examples" have been an extra data path in v1.44, and too much code was pasted over.

MiloradCvjetkovic, 2024-11-28:
Hi @thorstendb-ARM , 
is this available in the nightly build for testing, or is this still in development and will be available with next extension release?

MiloradCvjetkovic, 2024-12-03:
Thanks @thorstendb-ARM , I have tested and it works well.