ARM-software/vscode-cmsis-csolution #130: Create a New Solution does not copy/retain original `.cbuild-set.yml` from example

MiloradCvjetkovic, 2024-11-21:
Type: Bug Report

**Extension Version**: 1.44.0

**Description**

Using `Create a New Solution` button for copying example does not copy/retain original `.cbuild-set.yml` from example if such file already exists in the example, but the file gets created with only first context selected.

**Example**

`Fault.cbuild-set.yml` with the following content:
```
# Comment to see what happens with this file

cbuild-set:
  generated-by: vscode-cmsis-csolution version 1.44.0
  contexts:
    - context: Fault_S.Debug+B-U585I-IOT02A
    - context: Fault_NS.Debug+B-U585I-IOT02A
```

in the copied/created version looks like below:
```
cbuild-set:
  generated-by: vscode-cmsis-csolution version 1.44.0
  contexts:
    - context: Fault_S.Debug+STM32U585AIIx
```

**Expected behavior**

If the `.cbuild-set.yml` exists in the example it should be copied 1-to-1 to copy location and left untouched.

JonatanAntoni, 2024-11-27:
Since the `.cbuild-set.yml` is written by the extension this seems not to retain the content on workspace load.

closed 2024-12-02 by jkrech

MiloradCvjetkovic, 2024-12-03:
Hi @JonatanAntoni , 

I have now tested with development version of the extension (1.45), and I have seen that line `generated-by` is regenerated which I think it shouldn't be, the `.cbuild-set.yml` should be retained the way it is, as that `generated by` is indicative of how that file was initially created.

My `.cbuild-set.yml` containing:
`generated-by: vscode-cmsis-csolution version 1.44.0`

Has turned into:
`generated-by: vscode-cmsis-csolution version 1.45.0`

JonatanAntoni, 2024-12-03:
Probably, this update should only happen if the content is changed as well. I'll have a look.