ARM-software/vscode-cmsis-csolution #131: uv2csolution optimize not considered

minicarpet, 2024-11-22:
Type: Bug Report

**Describe the bug**
- OS and Version: Windows x64
- VS Code Version: 1.95.3
- Extension Version: 1.44.0
- Target Device: STMicroelectronics::STM32H523CEUx
- Other extensions you installed (and if the issue persists after disabling them):

Converting a .uvprojx does not consider the optimization

**To Reproduce**
Steps to reproduce the behavior:
1. Create a keil with optimization

[image]

3. Click on "Convert a uVision Project" and choose the project just created

[image]

4. See error (optimize yml should not be "debug")

[image]

> **_NOTE:_**  Optimize field [(yml description)](https://github.com/Open-CMSIS-Pack/cmsis-toolbox/blob/main/docs/YML-Input-Format.md#optimize) seems to not support all possible options.

**Expected behavior**
Optimize should be:
> debug if O0 or O1 selected
> speed if Ofast selected
> size if Oz image size selected
> balanced if Os balanced
No possible value for others values into keil ??

mguzmanm, 2024-11-25:
If "Debug Information" is set in "Output" tab ("Options for target" window in uVision), "Optimization" value in "C/C++ (AC6)" tab, will be overwritten with "debug" value. Please see requirement in https://jira.arm.com/browse/SDMDK-10727. Hence, this issue is not a bug.

closed 2024-11-25 by mguzmanm

jkrech, 2024-11-25:
The [optimize: yml node](https://github.com/Open-CMSIS-Pack/cmsis-toolbox/blob/main/docs/YML-Input-Format.md#optimize) deliberately chooses "generic" levels rather than toolchain specific command line flags. When mapping from uVision settings to the CMSIS Solution format multiple uVision values need to be mapped to a single YML value.

The unexpected behavior is triggered by the setting Options for Target -> Output Tab -> Debug Information.
This currently forces the optimization to "debug" to ensure best possible debug experience. Please uncheck that option in uVision to enable optimization.

**uVision:  => CMSIS Solution**
<default> =>  
-O0  => optimize: none
-O1 => optimize: none
-O2 => optimize: speed
-O3 => optimize: speed
-Ofast => optimize: speed
-Os balanced => optimize: balanced
-Oz image size => optimize: size
-Omax => optimize: speed

minicarpet, 2024-11-25:
-Omax seems to put balanced insead of speed, but it's fine for ma as I use -Ofast, thank you

chentang16, 2024-11-26:
> The [optimize: yml node](https://github.com/Open-CMSIS-Pack/cmsis-toolbox/blob/main/docs/YML-Input-Format.md#optimize) deliberately chooses "generic" levels rather than toolchain specific command line flags. When mapping from uVision settings to the CMSIS Solution format multiple uVision values need to be mapped to a single YML value.
> 
> The unexpected behavior is triggered by the setting Options for Target -> Output Tab -> Debug Information. This currently forces the optimization to "debug" to ensure best possible debug experience. Please uncheck that option in uVision to enable optimization.
> 
> **uVision: => CMSIS Solution** `<default>` => -O0 => optimize: none -O1 => optimize: none -O2 => optimize: speed -O3 => optimize: speed -Ofast => optimize: speed -Os balanced => optimize: balanced -Oz image size => optimize: size -Omax => optimize: balanced
> 
> Please note that there is always the possibility to specify toolchain specific command line options using the [misc:](https://github.com/Open-CMSIS-Pack/cmsis-toolbox/blob/main/docs/YML-Input-Format.md#misc) node in case the generic abstractions lack granularity.

In my opinion such a mapping between the "generic" levels and -O levels should be well documented somewhere. Users would instinctually assume the compiler options are one-to-one converted to the csolution project.