ARM-software/vscode-cmsis-csolution #132: Create solution from Blank_project template adds wrong board vendor

ReinhardKeil, 2024-11-27:
Details see https://github.com/Open-CMSIS-Pack/devtools/issues/1671

JonatanAntoni, 2024-11-27:
Solar web service provides the "wrong" board vendor name. While in the PDSC file ABOV uses just `ABOV` as pack and board vendor names, we forced them to use `ABOV Semiconductors` as device vendor. Solar uses the device vendor enum in all places. This problem is expected to be solved when getting all data from Core Tools directly, as Thorsten indicated.

JonatanAntoni, 2024-12-10:
Depends on #147

ReinhardKeil, 2025-01-14:
I suggest we implement for now:
  - Templates and Reference Apps are only shown from local installed packs.
  - Examples are shown from both web and local packs.

JonatanAntoni, 2025-01-20:
@ReinhardKeil, do you want the blank templates to be removed from the list for devices from web packs?

ReinhardKeil, 2025-01-21:
When a device comes from a web service, there should be only examples shown.  No templates, even no built-in templates should be listed.

JonatanAntoni, 2025-01-21:
@ReinhardKeil, I think the underlaying problem has already been "solved" by adding boards/devices only by name omitting the vendor's name. This is expected to work unless there is a name clash on board/device names between different vendors *and* both packs are selected for the solution. Such a clash is very unlikely to happen. Hence, I'd leave it as is and keep the blank templates also for web results.

DavidLesnjak, 2025-01-22:
@JonatanAntoni, adding a board without vendor name has a side effect on a cbridge. If a board doesn't have board vendor specified, cbridge can't launch STM32CubeMX to be preconfigured for a board.

DavidLesnjak, 2025-01-22:
There is no "clean" way to solve this in cbridge. Cbridge uses cbuild-gen-idx.yml as an input parameter.  I didn't analyze the order of commands., but I assume that cbuild-gen-idx.yml is generated when generator is launched by the user and at that point all packs are already installed. 
@brondani, could you maybe help with this? Is it possible that board vendor is add to the cbuild-gen-idx.yml although is missing in csolution.yml?

brondani, 2025-01-22:
> [@brondani](https://github.com/brondani), could you maybe help with this? Is it possible that board vendor is add to the cbuild-gen-idx.yml although is missing in csolution.yml?

Yes this will be fixed, tracked in https://github.com/Open-CMSIS-Pack/devtools/issues/1930

closed 2025-01-29 by jkrech