ARM-software/vscode-cmsis-csolution #136: Suggest to add `--packs` option also to build

ReinhardKeil, 2024-11-28:
Currently only `cbuild setup` uses `--packs` (if Download Packs option is enabled).

However, in case that new packs are published and `cpackget update-index` is used on the command line, the `cbuild setup` is not triggered and missing packs are not downloaded.

[image]

I suggest therefore to add also `--packs` to the build process itself.

JonatanAntoni, 2024-12-10:
Once watching the `pack.idx` file I'd expect a refresh is triggered, automatically. That should cover this case as well.

jkrech, 2024-12-10:
cpackget starting CMSIS-Toolbox 2.7.0 will implicitly update the index.pidx as part of `cpackget add` commands at least once a day. Therefore an externally triggered `cpackget update-index` is not expected to be a concern.
When loading a csolution.yml file `cbuild setup` will be executed and will always download latest available pack versions  (if online and no version is specified). Note that this will create a `*.cbuild-pack.yml` file and regardless of installed pack versions, these versions will be used.

JonatanAntoni, 2025-01-20:
What shall be done, here? Do we expect situations where `cbuild` is used to build a project without `cbuild setup` has run before?

jkrech, 2025-01-21:
@JonatanAntoni : While I cannot currently see a scenario, where there is a pack not installed yet when a build is triggered, I also cannot see circumstances where the `--packs` option would have an undesired side effect.
Reading the CMSIS Solution settings description for `Download Packs` it clearly states that this option will set the `--packs` option for both setup and build.
Therefore I suggest to add that command line option for build if the extension setting is enabled.

jkrech, 2025-01-28:
Internal build shows that `--packs` is now also present for the build action:
Enqueue: cbuild c:\workspace\testBlank\Blank_solution\Blank_solution.csolution.yml --context-set --packs
Execute: cbuild c:\workspace\testBlank\Blank_solution\Blank_solution.csolution.yml --context-set --packs

closed 2025-01-28 by jkrech