ARM-software/vscode-cmsis-csolution #145: Add "Help" to context menu

KeilChris, 2024-12-06:
Type: Feature Request

Clicking on the three dots in the CMSIS view shows a context menu. I would like to extend that context menu with a "Help" entry. Clicking on this entry should open a Markdown file in preview mode that I will provide (e.g. Help.MD).
Reasoning: the documentation is scattered across various places on the Arm web estate and we should create a central "go to page" that users can easily access.

Example:
[image]

ReinhardKeil, 2024-12-06:
ToDo: define link location

KeilChris, 2024-12-06:
> ToDo: define link location

The Markdown file would be provided with the extension.

mguzmanm, 2025-01-09:
Done.

Note: It was agreed to use arm developer help's link for the moment.

Arm developer help's link:
https://developer.arm.com/documentation/108029/0000/Get-started-with-an-example-project?lang=en

mguzmanm, 2025-01-24:
Link was updated.

mguzmanm, 2025-01-24:
Chris, please check help navigation, it seems that some links are not pointing out to the right index.html files.

closed 2025-01-27 by mguzmanm