ARM-software/vscode-cmsis-csolution #146: Pack Online Help does not open in Ubuntu

FerdinandEn, 2024-12-06:
Type: Bug Report

**Describe the bug**
- OS and Version: 
   Distributor ID:	Ubuntu
  Description:	Ubuntu 24.04.1 LTS
  Release:	24.04
  Codename:	noble

- VS Code Version: 1.95.3
- Extension Version: 1.44.0
- Target Device: STM32F756ZGTx

**To Reproduce**
Steps to reproduce the behavior:
1. open CMSIS View in VSC
2. Click on ARM::CMSIS:CORE -> Open Documentation
3. Firefox is opens with error message

[image]

**Expected behavior**
Help opens

jkrech, 2024-12-09:
@FerdinandEn, could you please check whether this file exists for you, and the file permissions of the file?
Does this problem exist for other documentation as well?

FerdinandEn, 2024-12-09:
@jkrech  Here are the screen shots of the pack directory and the index.html file.
I have tried the same with CMSIS-DSP. The same result. A colleague of mine has the same problem.

[image]

[image]

HansSchneebauer, 2025-01-24:
@FerdinandEn, I can duplicate this issue and have analyzed why it happens.
In Ubuntu, the web browsers like Firefox or Chromium are installed as a Snap package by default. This means that these applications run in a sandbox with limited access rights because of security reasons. One of the limits is that no local files can be accessed that are located in a hidden (sub-) directory. Since all CMSIS software packs are installed by default into the path ${HOME}/.cache/arm/packs, no documentation file of a software pack can be read by a web browser.
I see two different workarounds for now:

**1. Use a different path for the CMSIS software packs** (preferred one)
You just need to set the environment variable CMSIS_PACK_ROOT to a path that does not include a hidden folder (=starting with a dot). I tried with  /home/<user>/CMSIS-Packs and opening CMSIS-pack related documentation from VS Code worked as expected.
**2. Use a Web Browser that is not installed as a Snap package**
You could download Firefox from the Mozilla website, uninstall the Firefox that was pre-installed with Ubuntu, and then install the one you downloaded. When you install Firefox this way, it does not run in a sandbox and CMSIS-pack related documentation can be viewed as expected.

jkrech, 2025-02-13:
We will not change the default location of the CMSIS Pack repository. We now ship documentation as part of the extension e.g.
`C:\Users\<username>\.vscode\extensions\arm.cmsis-csolution-1.48.0-win32-x64\guide falling into the same category.
As mitigation we will add this to a new `Troubleshooting` section.

Web Browser installed as Snap package do not allow to open files from a directory with a folder with a leading `.` (.../.folder/`).
Install and configure a Web Browser that is not installed as a Snap package as default browser.

jkrech, 2025-06-30:
https://mdk-packs.github.io/vscode-cmsis-solution-docs/tipsandtricks.html#documentation-does-not-open

closed 2025-06-30 by jkrech