ARM-software/vscode-cmsis-csolution #147: Open Issues with Web Service

ReinhardKeil, 2024-12-09:
The CMSIS Solution obtains information about Boards and Devices via Web Services.  This workflow downloads packs seamless, but has currently a few flaws:

- Device specific templates are not listed, when the DFP is not installed on the local computer. Instead the build-in default templates are shown.
- Reference examples are not shown when the BSP is not installed on the local computer (as there is not information about layers in the BSP obtained by the Web Service.

jkrech, 2024-12-09:
This issue https://github.com/ARM-software/vscode-cmsis-csolution/issues/117 is related

ReinhardKeil, 2025-02-04:
I believe the workarounds are OK.  IMHO it is sufficient to just list examples for BSPs that are not installed.

ReinhardKeil, 2025-02-05:
ticket is parked for future enhancements - currently no action planned