ARM-software/vscode-cmsis-csolution #148: Show "P" for files that are located in packs

ReinhardKeil, 2024-12-09:
The CMSIS view shows on the right side symbols such "U" (for Untracked).

Is it possible to add "P" (Pack sourced) for files that are coming from {CMSIS_PACK_ROOT}. This would indicated to the user that these files should not be modified.

[image]

maribelguzmanm, 2025-01-08:
Symbol "U" comes from Git. Adding a new Git file status to the Git API within a VS Code extension is limited to the Git statuses (Untracked, Tracked, Modified, Staged, Deleted, Renamed, Copied, Ignored...) that Git itself supports. However, is possible to add a custom file status by e.g.:  using file decorations, adding a lock icon or adding file description.

ReinhardKeil, 2025-01-15:
The text to show is: "\<dot\> Pack sourced"

[image]

Similar to this:

[image]

closed 2025-01-16 by mguzmanm