ARM-software/vscode-cmsis-csolution #149: "Add Software Layer": Check if path already exists

thorstendb-ARM, 2024-12-09:
Type: Feature Request

The layer dialog has another problem. If the destination directory exists already, the files are overwritten.

Potential solution:

Output in RED: The chosen sub-directory "ABC" already exists. // this is similar to Create New Solution dialog
We could also compare the content of both directories and when identical don't show above message.

See https://github.com/ARM-software/vscode-cmsis-csolution/issues/137

thorstendb-ARM, 2025-01-14:
Added feature to https://github.com/ARM-software/vscode-cmsis-csolution/issues/149
[image]

jkrech, 2025-01-31:
Validated in internal pre-release 1.47: "Path already exists" is displayed as shown in screenshot above. No further testing is done whether the existing content is different from the new layer folder to be copied.

closed 2025-01-31 by jkrech