ARM-software/vscode-cmsis-csolution #150: [Feature] Discuss whether Action Icons like e.g. "launch generator", "open documentation" should be displayed permanently in the "CMSIS Outline View"

jkrech, 2024-12-10:
currently action buttons for running a generator:
[image]

or opening the documentation (book icon) are displayed only on selection of the component or hover over.

If technically feasible, displaying e.g. example the generator launch icon could benefit from permanent display for more easily identifying generator components within the CMSIS outline view.

ReinhardKeil, 2024-12-19:
Implement a permanent display of the "Generator" button

[image]

jkrech, 2025-01-23:
See: https://github.com/ARM-software/vscode-cmsis-csolution/issues/167

closed 2025-01-23 by jkrech