ARM-software/vscode-cmsis-csolution #153: UX Improvement: When only one compiler can be choose, select it automatically

ReinhardKeil, 2024-12-12:
When there is only one choice for compiler, select the compiler without user interaction and just output a notification.

This would avoid that such a dialog is shown:

[image]

thorstendb-ARM, 2024-12-12:
See PR https://github.com/Arm-Debug/vscode-cmsis-csolution/pull/2020

jkrech, 2024-12-16:
Validated using internal build of CMSIS Solution extension from main branch 2024-12-16 8:00.
a) only one toolchain installed (AC6) -> "Configure Solution" Editor View does not open and "cbuild setup" immediately completes without further user interactsion
b) when adding a second toolchain (AC6, GCC) -> "Configure solution" Editor View opens with Compiler Selection for GCC and AC6.

closed 2024-12-16 by jkrech