ARM-software/vscode-cmsis-csolution #154: UX Improvement: Hide build output in .cmsis folder

ReinhardKeil, 2024-12-12:
Type: Feature Request

Consider to add an option to hide output files in the `.cmsis` folder.   This could be a configurable option that enables `--output` for the cbuild command.

thorstendb-ARM, 2025-01-27:
Implemented, see PR: https://github.com/Arm-Debug/vscode-cmsis-csolution/pull/2084

jkrech, 2025-01-29:
Validated internal build of the extension. 
Experimented with setting using `.cmsis` and all generated *.cbuild.yml and tmp and out folders were created in that path as expected.
`--output <path>/.cmsis` have been added to all `cbuild` invocations.
Any further impact for features accessing the cbuild files need to be investigated in more detail.