ARM-software/vscode-cmsis-csolution #155: Change links to new CMSIS-Toolbox documentation

ReinhardKeil, 2024-12-13:
Replace documentation links like this:
"https://github.com/Open-CMSIS-Pack/cmsis-toolbox/blob/main/docs/YML-Input-Format.md"
With:
"https://open-cmsis-pack.github.io/cmsis-toolbox/YML-Input-Format"

thorstendb-ARM, 2025-01-14:
Done:
`- [File groups](https://open-cmsis-pack.github.io/cmsis-toolbox/YML-Input-Format#groups): Groups used to organize user source code.
`

Elise-Kaminski, 2025-01-14:
Documentation PRs:

- VS Code extensions UG: https://github.com/Arm-Debug/ks-vscode-extension-docs/pull/283

- KSC Theia UG: https://github.com/Arm-Debug/mbed-studio-docs/pull/310

- MDK GSG: https://github.com/Arm-Debug/mdk-docs/pull/86

- CMSIS Solution Readme: https://github.com/Arm-Debug/vscode-cmsis-csolution/pull/2052

thorstendb-ARM, 2025-01-14:
Exchanged and tested all links found in CSolution source.

jkrech, 2025-01-31:
Links to CMSIS-Toolbox documentation have been updated to new location.

closed 2025-01-31 by jkrech