ARM-software/vscode-cmsis-csolution #157: ERROR:1 CANCELLED 1 CANCELLED: Call cancelled

Mcreepers, 2024-12-15:
Type: Bug Report

**Describe the bug**
- OS and Version:Ubuntu24.04
- VS Code Version:Vscode1.59.1
- Extension Version:Arm CMSIS Solution v1.44.0
- Target Device:STM32G473VET6
- Other extensions:ARM Debugger v1.4.3
- Description:When I tried to flash device after build all,but flash failed,I get the following error message:
`1 CANCELLED 1 CANCELLED: Call cancelled`

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'CMSIS'
2. Click on 'RUN'
3. See error

**Expected behavior**

Flash success.

**Code sample and logs**
- `launch.json`
```
{
	"version": "0.2.0",
	"configurations": [
		{
			"name": "Arm Debugger",
			"type": "arm-debugger",
			"request": "launch",
			"serialNumber": "${command:device-manager.getSerialNumber}",
			"program": "${command:arm-debugger.getApplicationFile}",
			"cmsisPack": "${command:cmsis-csolution.getTargetPack}",
			"deviceName": "${command:cmsis-csolution.getDeviceName}",
			"processorName": "${command:cmsis-csolution.getProcessorName}"
		}
	]
}
```
- vscode logs
I found logs in console below:
```
console.ts:137 [Extension Host] Failed to resolve hardware GrpcServiceError: 1 CANCELLED 1 CANCELLED: Call cancelled
    at Object.create (/home/binzs/.vscode/extensions/arm.cmsis-csolution-1.44.0-linux-x64/dist/desktop/extension.js:2:1583357)
    at Object.parse (/home/binzs/.vscode/extensions/arm.cmsis-csolution-1.44.0-linux-x64/dist/desktop/extension.js:2:1583504)
    at a.<anonymous> (/home/binzs/.vscode/extensions/arm.cmsis-csolution-1.44.0-linux-x64/dist/desktop/extension.js:2:1581234)
    at a.emit (node:events:519:28)
    at Object.onReceiveStatus (/home/binzs/.vscode/extensions/arm.cmsis-csolution-1.44.0-linux-x64/dist/desktop/extension.js:2:574988)
    at Object.onReceiveStatus (/home/binzs/.vscode/extensions/arm.cmsis-csolution-1.44.0-linux-x64/dist/desktop/extension.js:2:567953)
    at /home/binzs/.vscode/extensions/arm.cmsis-csolution-1.44.0-linux-x64/dist/desktop/extension.js:2:661126
    at processTicksAndRejections (node:internal/process/task_queues:77:11)
```
```
1 CANCELLED 1 CANCELLED: Call cancelled      notificationsAlerts.ts:42  
ERR 1 CANCELLED 1 CANCELLED: Call cancelled: Error: 1 CANCELLED 1 CANCELLED: Call cancelled        log.ts:439
    at Y0.h (file:///snap/code/174/usr/share/code/resources/app/out/vs/workbench/api/node/extensionHostProcess.js:114:33180)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
```

Usually this error can fix after install `libncurses.so.5`,but now it don't work.I know this solvetion when I use arm-edbi-none-gdb,I get error messge and I fixed.
I cannot get any information why it failed in error message:1 CANCELLED 1 CANCELLED: Call cancelled,it should show what dependency issues happend.

jkrech, 2024-12-16:
@Mcreepers, could you uplease share your file content in "task.json". Please note that the "run" command is executing the flash operation using the task.json - see task of type: ` "type": "arm-debugger.flash"`

Mcreepers, 2024-12-16:
Here,I use default "tasks.json".
```
{
	"version": "2.0.0",
	"tasks": [
		{
			"type": "arm-debugger.flash",
			"serialNumber": "${command:device-manager.getSerialNumber}",
			"program": "${command:arm-debugger.getApplicationFile}",
			"cmsisPack": "${command:cmsis-csolution.getTargetPack}",
			"deviceName": "${command:cmsis-csolution.getDeviceName}",
			"processorName": "${command:cmsis-csolution.getProcessorName}",
			"problemMatcher": [],
			"label": "arm-debugger.flash: Flash Device"
		}
	]
}
```

JonatanAntoni, 2024-12-16:
The implementation is about to change fundamentally with the upcoming release 1.46.0
@Mcreepers, please could you recheck this issue at your ende once the release is available and let us know if your problems are solved. If not, we need to investigate further.

omarArm, 2024-12-18:
Hi @Mcreepers , I would like your input on two questions please,
1. which debug probe are you using?
2. could you please add and run the following task into your tasks.json file? this will help us know where the issue is exactly
```
{
            "label": "TEST cmsis-csolution get functions",
            "type": "shell",
            "command": "echo 'Serial: ${command:device-manager.getSerialNumber}\nProgram: ${command:arm-debugger.getApplicationFile}\nPack: ${command:cmsis-csolution.getTargetPack}\nDevice: ${command:cmsis-csolution.getDeviceName}\nProcessor: ${command:cmsis-csolution.getProcessorName}'",
            "problemMatcher": []
        }
```

Mcreepers, 2024-12-18:
@omarArm Actually, the output of this task is the same as the previous one, and I can't find any other output
[image]

omarArm, 2024-12-18:
@Mcreepers 
oh it seems you reran the old task, instead press on ctrl+shift+P and choose the task with the name "Test get functions". I should have clarified that in the past commaned

omarArm, 2024-12-19:
Hi @Mcreepers ,
I would like to share that a new version of the extension is now released, can you please double check again your project with the new version 1.46.0?

Mcreepers, 2024-12-21:
@omarArm the issue has been resolved after upgrading to version 1.46.0. Now, when I execute the task you provided, I am able to receive output in the terminal. However, I must clarify that prior to running the task you provided, I was unable to obtain any output except for errors.If you don't have anything else to do, I'm going to close this issue.
[image]

omarArm, 2024-12-23:
I am glad the issue got resolved with 1.46.0, no more input now

closed 2024-12-23 by Mcreepers