ARM-software/vscode-cmsis-csolution #162: [Solution invalid] when offline

ReinhardKeil, 2024-12-19:
VSCode CMSIS Csolution extension 1.46
CMSIS_Toolbox 2.7
OS: Windows

When the internet connection is missing, `[Solution invalid]` is displayed in the status bar.
Side effects are: Manage Solution view cannot be opened.

Test-Example: MDK-Middleware USBD.csolution.yml

[image]

JonatanAntoni, 2025-01-16:
I cannot reproduce this issue. Perhaps VSCode downloads the schema and caches it somewhere locally on first use. But even if, the `[Solution invalid]` is shown by our code if an exception/error occurs while loading the file. I don't see VSCode schema checking involved at this point. The schema checking/assistance should only kick in once a file is opend in an editor window.

Instead of referring to online schemas, it should work to refer to a local file shipped with the extension. E.g., the Markdownlint extension does it this way: https://github.com/DavidAnson/vscode-markdownlint/blob/159fdca598db4b4fb8b6dd212a87efebaec7adc7/package.json#L131-L143

jkrech, 2025-01-28:
Validating internal pre-release 1.47:
updated Schema files version 2.7.1 shipped as part of the extension - no dependency on online connection anymore.

closed 2025-01-28 by jkrech