ARM-software/vscode-cmsis-csolution #164: STM32H7RS project creation is not working

FerdinandEn, 2024-12-21:
Bug Report

- OS and Version: Windows 11
- VS Code Version: 1.96.2
- CMSIS-Solution-Extension 1.46.0
- CMSIS-Soultion 2.7.0

With the current version of the tools, it is not possible to create a project for the STM32H7RS. This processor has a boot and an application project. 
The two projects are not created. I have created both projects manually. You can start CubeMX and the files are also generated by CubeMX. However, they are not integrated into the projects.

- No two projects are offered when the project is created.  For a multicore (CM7/CM4) or Trustzone project, the two required projects are offered.
- The cgen files required for integrating the CubeMX files are not created

**To Reproduce**
Steps to reproduce the behavior:
1. Create a Project for STM32H7S3L8 -> Only one project
2. Include CubeMX  (Software Components)
3. Start CubeMX and generate Code -> cgen not created

The same applies to the Keil MDK (uVision)

ReinhardKeil, 2024-12-27:
Thanks for raising this issue. We are working with ST on a solution. Expect it in Feb/Mar 2025 (as it requires a change in CubeMX).

JonatanAntoni, 2025-01-28:
Reproduced: In v1.46.0 and the device `STM32H7S3L8Hx` only `Blank solution` template is shown. This default template needs manual project configuration and seems not to work for CubeMX.

JonatanAntoni, 2025-01-29:
Second issue is the outdated/erroneous Cube template shipped in STM32H7RSxx_DFP v1.0.0. The `.cproject.yml` files are missing the following snippet:

```yaml
  generators:
    options:
    - generator: CubeMX
      map: <project>
```

This issue is solved in https://github.com/Open-CMSIS-Pack/STM32H7RSxx_DFP/blob/b801921a3c695749efbd408a061ddf5e45d214c3/Templates/CubeMX/Appli/Appli.cproject.yml#L10C1-L13C17 and will be shipped in the next release of the DFP.

As a workaround, one could either patch the `.cproject.yml` files manually, or install the `dev` branch of https://github.com/Open-CMSIS-Pack/STM32H7RSxx_DFP as a local repository:

```
gh clone Open-CMSIS-Pack/STM32H7RSxx_DFP
cd STM32H7RSxx_DFP
git checkout dev
cpackget add Keil.STM32H7RSxx_DFP.pdsc
```

**Attention:** cpackget 2.1.5 which is shipped in CMSIS-Toolbox 2.7.0 is affected by Open-CMSIS-Pack#421. The above use of `cpackget add` to register the local pack will corrupt `local_repository.pidx` file so that it gets ignored and no local packs are used anymore.

jkrech, 2025-04-04:
UPDATE: 2025-04-04
CMSIS Solution extension 1.50
CMSIS-Toolbox 2.8.0

Install:
- STM32CubeMX 6.14 is available from ST.com
- latest Firmware Package: STM32CubeH7RS Firmware Package V1.2.0
- Keil::STM32H7RSxx_DFP@1.0.1 is published
- Keil::STM32H7S78-DK_BSP@1.0.0 is published

Create new solution, select 
board: STM32H7S78-DK (Rev.D)
project: CubeMX Basic Solution

Launch Generator from the CMSIS View displayed for component `Device:CubeMX`
Run "Generate Code" in STM32CubeMX

`cbuild setup` and `cbuild` execute successfully for me.

closed 2025-04-08 by jkrech