ARM-software/vscode-cmsis-csolution #167: UX Improvement: Make "Run Generator" a permanent action button

ReinhardKeil, 2025-01-09:
The components that start a generator are "special". To make them stand out better, make the action button "Run Generator" permanent.  Currently it is only shown at "Hover over".

[image]

edriouk, 2025-04-04:
ChatGPT belives it is not supported: 
the VS Code API for TreeView, action buttons (aka inline actions or TreeItem "buttons") are typically defined using the TreeItem.command or TreeItem.contextValue combined with a when clause in the contributes.menus section of your package.json under view/item/context.

However, by default, inline actions in TreeViews only appear on hover — and VS Code does not currently support making them always visible via the API.

**Workaround / Alternative Approaches**
1. Use Labels or Icons as Static UI
If you want to simulate a "button" that's always visible:
Embed an icon in the label (e.g., with a special character or emoji)
Use a TreeItem.command so clicking the item performs an action
But this isn't interactive like an inline button and only allows a single command per item.

2. Use TreeItem.resourceUri and decorations
You can apply icons via file decorations, but again, not clickable.

3. File a Feature Request
Many developers have requested this feature:

**Summary**
Unfortunately, making inline action buttons always visible in a TreeView is not currently supported by the VS Code API. You can:

- Use the hover-based context actions (default behavior)
- Simulate a button via label icons and commands
- Monitor for API updates or file an issue with VS Code

mguzmanm, 2025-04-04:
Option 1 is also not feasible since components may have a book icon as well.

[image]