ARM-software/vscode-cmsis-csolution #169: downgrading compiler version with Arm Tools Environment Manager not working

andbararm, 2025-01-09:
Type: Bug Report

**Describe the bug**
- OS and Version: tested on Windows 11
- VS Code Version: 1.96.2
- Extension Version: 1.46.0
- Arm Tools Environment Manager: 1.10.0

The expected procedure (given in Arm Tools Environment Manager Settings "Reactivate On Config Change" is enabled) to change used Arm Compiler version is to change the vcpkg-configuration.json file, save it and do a "Rebuild Solution". However this only works when upgrading the compiler, like from 6.22 to 6.23. This did not work for downgrading the compiler like from 6.23 to 6.22. As a workaround one can execute from the Command Palette the command "Developer: Reload Window" and then doing a "Rebuild Solution".

Steps to reproduce the behavior:
1. Go to 'vcpkg-configuration.json'
2. Downgrade the compiler version from like 6.23 to 6.22 and save the file.
3. Do a "Rebuild Solution"
4. Problem: in compile_commands.json still the version 6.23 is in use.

**Expected behavior**

The compile_commands.json should show that the version 6.22 is in use.

jkrech, 2025-01-13:
This is most likely an issue with the CMSIS-Toolbox cbuild setup command.

ReinhardKeil, 2025-01-16:
I believe a workaround is to restart VSCode

JonatanAntoni, 2025-01-21:
> I believe a workaround is to restart VSCode

Deactivating the environment explicitly and reactivate it cleanly does the job. One doesn't need to restart VSCode as a whole.

JonatanAntoni, 2025-02-06:
Filed Arm-Debug/vscode-environment-manager#175. We can't do anything about it in CMSIS Solution extension.

jkrech, 2025-04-24:
Even using a more recent vcpkg version I still see issues.

JonatanAntoni, 2025-04-24:
> Even using a more recent vcpkg version I still see issues.

The problem is not in vcpkg itself, but in the Arm Environment Manager way of integrating it into VS Code. Using `vcpkg-shell` on a terminal doesn't show this issue.

jkrech, 2025-10-01:
@KeilChris this problem does not seem to get fixed anytime soon.
The problem is that the environment variables for the toolchain will be added for each new version but not removed until you 
a) Deactivate Environment
b) Activate Environment

Can we add this to the tips and tricks section: https://mdk-packs.github.io/vscode-cmsis-solution-docs/tipsandtricks.html