ARM-software/vscode-cmsis-csolution #170: Crashes when compile converted project for GD32 chip

Aptahar, 2025-01-10:
Type: Bug Report

- OS and Version: Ubuntu 24.10
- VS Code Version: 1.96.2
- Extension Version: 1.46.0
- Target Device: GD32F427RK

There is a uvprojx project that works in Keil uVision.
But when convert it to use in VSCode and start compiling, then VScode crashes.

uVision project and VSCode logs:
[keil_project.zip](https://github.com/user-attachments/files/18371655/keil_project.zip)
[VSCode_logs.zip](https://github.com/user-attachments/files/18371681/VSCode_logs.zip)

jkrech, 2025-01-17:
I tried to reproduce the crash but had only a Windows machine to do so. I was successful in building the converted csolution project. Are you able to successfully build the project if you invoke the build from the terminal?
```
  cbuild M101SN.csolution.yml -S
```

Aptahar, 2025-01-17:
> I tried to reproduce the crash but had only a Windows machine to do so. I was successful in building the converted csolution project. Are you able to successfully build the project if you invoke the build from the terminal?
> 
> ```
>   cbuild M101SN.csolution.yml -S
> ```

It also crash VSCode

JonatanAntoni, 2025-01-21:
@Aptahar, I tried the content provided in keil_project.zip converted using uv2csolution 1.5.0 shipped in Arm CMSIS Solution extension 1.46.0 on my Mac without issues.

Unfortunately, your log archive doesn't contain any useful output. Could you please rerun all your steps, log them detailed and provide all output?

I read your initial request correctly, the conversion step is working and you can load the converted csolution into VSCode? Can you please provide the conversion result, i.e. zip archive the project folder after running uv2csolution?

Then, you could try to run the build outside of vs code. Just open the project folder in a terminal Window. In order to get access to all the tools you might need to activate the vcpkg configuration within the external terminal. Then run `cbuild M101SN.csolution.yml -S` command to build.

```
❯ vcpkg-shell activate
warning: vcpkg-artifacts is experimental and may change at any time.
Artifact                                Version Status    Dependency Summary
arm:tools/kitware/cmake                 3.28.4  installed            Kitware's cmake tool
arm:tools/ninja-build/ninja             1.12.0  installed            Ninja Build
arm:compilers/arm/armclang              6.23.0  installed            Arm Compiler for Embedded
arm:tools/open-cmsis-pack/cmsis-toolbox 2.7.0   installed            Arm CMSIS-Toolbox

Activating: ./m101sn

❯ cbuild M101SN.csolution.yml -S

+-------------------------------------
(1/1) Building context: "M101SN+Debug"
Building CMake target 'M101SN+Debug'
[0/1] Re-running CMake...
-- Configuring done (0.2s)
-- Generating done (0.0s)
-- Build files have been written to: /m101sn/tmp/1
[1/20] Building C object CMakeFiles/GigaDevice_Device_StdPeripherals_CRC_3_0_2.dir/m101sn/RTE/Device/GD32F427RK/gd32f4xx_crc.o
[2/20] Building C object CMakeFiles/Group_src.dir/m101sn/src/init.o
[3/20] Building C object CMakeFiles/Group_src.dir/m101sn/src/systick.o
[4/20] Building C object CMakeFiles/Group_system.dir/m101sn/src/system_gd32f4xx.o
[5/20] Building C object CMakeFiles/GigaDevice_Device_StdPeripherals_EXTI_3_0_2.dir/m101sn/RTE/Device/GD32F427RK/gd32f4xx_exti.o
[6/20] Building C object CMakeFiles/GigaDevice_Device_StdPeripherals_FWDGT_3_0_2.dir/m101sn/RTE/Device/GD32F427RK/gd32f4xx_fwdgt.o
[7/20] Building C object CMakeFiles/GigaDevice_Device_StdPeripherals_GPIO_3_0_2.dir/m101sn/RTE/Device/GD32F427RK/gd32f4xx_gpio.o
[8/20] Building C object CMakeFiles/GigaDevice_Device_StdPeripherals_MISC_3_0_2.dir/m101sn/RTE/Device/GD32F427RK/gd32f4xx_misc.o
[9/20] Building C object CMakeFiles/GigaDevice_Device_StdPeripherals_ADC_3_0_2.dir/m101sn/RTE/Device/GD32F427RK/gd32f4xx_adc.o
[10/20] Building C object CMakeFiles/GigaDevice_Device_StdPeripherals_DMA_3_0_2.dir/m101sn/RTE/Device/GD32F427RK/gd32f4xx_dma.o
[11/20] Building C object CMakeFiles/GigaDevice_Device_StdPeripherals_FMC_3_0_2.dir/m101sn/RTE/Device/GD32F427RK/gd32f4xx_fmc.o
[12/20] Building C object CMakeFiles/GigaDevice_Device_StdPeripherals_ENET_3_0_2.dir/m101sn/RTE/Device/GD32F427RK/gd32f4xx_enet.o
[13/20] Building C object CMakeFiles/GigaDevice_Device_StdPeripherals_PMU_3_0_2.dir/m101sn/RTE/Device/GD32F427RK/gd32f4xx_pmu.o
[14/20] Building C object CMakeFiles/GigaDevice_Device_StdPeripherals_SYSCFG_3_0_2.dir/m101sn/RTE/Device/GD32F427RK/gd32f4xx_syscfg.o
[15/20] Building C object CMakeFiles/GigaDevice_Device_StdPeripherals_SPI_I2S_3_0_2.dir/m101sn/RTE/Device/GD32F427RK/gd32f4xx_spi.o
[16/20] Building C object CMakeFiles/GigaDevice_Device_StdPeripherals_RCU_3_0_2.dir/m101sn/RTE/Device/GD32F427RK/gd32f4xx_rcu.o
[17/20] Building C object CMakeFiles/GigaDevice_Device_StdPeripherals_USART_3_0_2.dir/m101sn/RTE/Device/GD32F427RK/gd32f4xx_usart.o
[18/20] Building C object CMakeFiles/GigaDevice_Device_StdPeripherals_TIMER_3_0_2.dir/m101sn/RTE/Device/GD32F427RK/gd32f4xx_timer.o
[19/20] Linking C executable /m101sn/out/M101SN/Debug/M101SN.axf
Program Size: Code=1404 RO-data=424 RW-data=4 ZI-data=2156
+------------------------------------------------------------
Build summary: 1 succeeded, 0 failed - Time Elapsed: 00:00:01
+============================================================
```

You can also run the conversion on command line:
```
❯ vcpkg-shell use arm:tools/arm/mdk-toolbox
warning: vcpkg-artifacts is experimental and may change at any time.
Artifact                  Version Status    Dependency Summary
arm:tools/arm/mdk-toolbox 1.0.0   installed            Arm MDK-Toolbox

arm:tools/arm/mdk-toolbox - Documentation: https://github.com/MDK-Packs/Documentation/tree/master/MDK-Toolbox#readme
Activating: arm:tools/arm/mdk-toolbox

❯ uv2csolution M101SN.uvprojx
1 out of 1 target(s) converted
Created solution: M101SN.csolution.yml
With projects:
- M101SN.cproject.yml
```

ReinhardKeil, 2025-02-04:
closing this issue as we cannot reproduce it.

closed 2025-02-04 by ReinhardKeil