ARM-software/vscode-cmsis-csolution #171: New solution workflow: Target type name ignored

KeilChris, 2025-01-14:
Type: Bug Report

**Description**

- OS and Version: MacOS 15.2.0
- VS Code Version: 1.96.3
- Extension Version: 1.46.0
- Target Device: B-U585-IOT02A

I have created a new CMSIS solution for the B-U585-IOT02A development board, based on the Blinky csolution example. I have entered a "Target Type" name for my target (MyHardware), but this name was ignored. In the generated csolution.yml file, I had this entry:
```yml
  # List different hardware targets that are used to deploy the solution.
  target-types:
    - type: B-U585I-IOT02A
      board: STMicroelectronics::B-U585I-IOT02A
      device: STMicroelectronics::STM32U585AIIx
```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Create new solution'.
2. 
3. Select B-U585-IOT02A.
4. Select "Blinky" Csolution example.
5. Enter "MyHardware" as **Target Type** (instead of STM32U585AIIx).
6. Click **Create**.

**Expected behavior**
The Blinky.csolution.yml file should show:
```yml
  # List different hardware targets that are used to deploy the solution.
  target-types:
    - type: MyHardware
      board: STMicroelectronics::B-U585I-IOT02A
      device: STMicroelectronics::STM32U585AIIx
```

ReinhardKeil, 2025-01-14:
The problem is that entry boxes like this should be grey when an example is selected.  Changing a target-type name in an example may create side-effects as the project may contain `for-context:` settings. 

For examples, entry fields like "Target Type" should be grey to indicated that these cannot change.

ReinhardKeil, 2025-01-15:
"Target Type" can only be entered for Templates.  By default it should be grey.

thorstendb-ARM, 2025-01-15:
Implemented here: https://github.com/Arm-Debug/vscode-cmsis-csolution/pull/2060
Edit "Target Type" now only shown on Templates (internal and DataManager).

jkrech, 2025-01-28:
Using internal pre-release 1.47 for validation:
a) after a built-in template has been selected the target-type name shows and can be altered.
b) for examples target-type is not displayed and cannot be altered.
c) for custom templates the target-type name is displayed but setting will be ignored
This does not seem to be expected.

closed 2025-01-29 by KeilChris