ARM-software/vscode-cmsis-csolution #172: Change/Add Features

ReinhardKeil, 2025-01-17:
Add the following commands to "Features"

ID                                        | Titel                |  Menu Contexts                  | Returns or Notes
:-----------------------------|:----------------|:------------------------------|:------------------------
cmsis-solution.GetDfpName  |  Name of Device Family Pack      |      commandPalette   | build:  device-pack: ARM::V2M_MPS3_SSE_300_BSP@1.5.0
cmsis-solution.GetBspName  |  Name of Board Support Pack     |      commandPalette  |  build:  board-pack: ARM::V2M_MPS3_SSE_300_BSP@1.5.0
cmsis-solution.GetDfpPath     |  Path to Device Family Pack      |      commandPalette   |  build:  absolute path of device-pack: ARM::V2M_MPS3_SSE_300_BSP@1.5.0
cmsis-solution.GetBspPath     |  Path to Board Support Pack     |      commandPalette   | build:  absolute path of board-pack:

thorstendb-ARM, 2025-01-20:
Note: All commands currently available are named csolution:
`"command": "cmsis-csolution.doSomething",`

thorstendb-ARM, 2025-01-24:
Section 'Add the following commands to "Features" ' fulfilled by merged PR and is ready for release.
https://github.com/Arm-Debug/vscode-cmsis-csolution/pull/2068

ReinhardKeil, 2025-02-03:
Remaining problems captured in #188